bigtreetech/SKR-3 #1: SKR 3 EZ

Dviros, 2022-06-18:
Dear BTT team,
Can you please also upload a klipper config for the EZ variant (I saw some differences in the schematics)?
I want to make sure I've set my config right.
Also, I'm working on getting it working in my Artillery Genius - you can check out my config here:
https://github.com/Dviros/Klipper-Configs/blob/main/skr_3_EZ_printer.cfg

Thanks in advance!

danikdd, 2022-07-06:
Hello Dviros, any progress here? My new SKR 3 EZ on the way ... cant find klipper config ... Its the same as SKR 3? Many thx for info. Daniel

Dviros, 2022-07-07:
Hey @danikdd , can confirm there are no specific changes between the 2.
You can check out mine - got everything running pretty damn good

danikdd, 2022-07-07:
ManyTHX for info, happy printing!

gitxmax, 2022-07-08:
> Hey @danikdd , can confirm there are no specific changes between the 2. You can check out mine - got everything running pretty damn good

Would u mind sharing the correct bin building options for klipper?

Dviros, 2022-07-09:
@gitxmax I've skipped the bootloader in the options in order to build and flash in DFU instead of the SD card flash, as this specific chip is not supposed tk be flashed using USB.
However, please note that
1. Couldn't find a way to revert to Marlin
2. I'm not sure on how to reflash the bootloader.

Will share the exact steps soon

Dviros, 2022-07-09:
[image]

`dfu-util -a 0 -s 0x8000000:leave -D out/firmware.bin`

@gitxmax

AfroPsycho, 2022-07-12:
> [image]
> 
> I have a typo; clock reference should be set to 25Mhz
> 
> `dfu-util -a 0 -s 0x8000000:leave -D out/firmware.bin`
> 
> @gitxmax

I guess you shouldn’t skip bootloader 
it is also mentioned in klipper guide 
I an using the klipper with [these](https://github.com/bigtreetech/SKR-3/tree/master/Firmware/Klipper)  settings as guided by btt

```
Microcontroller -> STM32
Processor -> STM32H743
Bootloader -> 128 kib bootloader ( SKR SE BX V2.0 )
Clock -> 25 MHz
Communication -> USART1 ( this depends on if you want to use usb or serial pin to communicate )
```
I assume selecting wrong clock speed could actually hit the performance of the board 
Also not selecting bootloader as you mentioned could result in not being able to use marlin again

Dviros, 2022-07-12:
Fair point, that PR was merged into Klipper not too long ago