bigtreetech/SKR-3 #3: Can’t use E0DET and E1DET when using sensorless homing on X and Y

Tony-g80, 2022-07-06:
I have enabled sensorless homing on my X and Y axes in marlin and all works well until I need to use the available spare pins of E0DET and E1DET, because these pins are defined as x/y/z max end stops they cant be used for filament detection function or even as spare pins if sensorless homing is enabled. 

If I did want to use the filament detection and sensorless homing together I wouldn’t be able to so it defeats the object of having the two very usable functions if you have to sacrifice one for the other

I have tried to work around this but it’s just not allowing it and I need to use those extra pins of E0/E1DET along with sensorless homing on x and y

Maybe one day soon I’ll be able to actually print something with this board once it all works 100%

radek8, 2022-07-08:
On the E1 and E0 controllers, just cut the pins in the DIAG connector and everything works as it should :-)

radek8, 2022-07-08:
So you don't have to cut anything.
Just remove the Jumper at the respective motor controller

[image]

Tony-g80, 2022-07-11:
I have the diag jumpers in place on the skr3 ez for x and y and it automatically reassigns the x/y max endstops in the pins to use the E0/E1DET pins which seems to cause issues for me using those pins for other things so if I want to use sensorless homing then it seems that I can’t use E0/E1DET as they become x/y max endstops, even though they are commented out in marlin config.h and not used it won’t allow me to use them if I’m using sensorless homing.

radek8, 2022-07-11:
If you remove the diag jumper under the engine E0 and E1,
you can use connectors E0DET and E1DET according to your configuration.
They are not physically configured to X and Y max.

Tony-g80, 2022-07-11:
I generally think your not understanding what I’m describing, I HAVE the diag jumpers connected for x and y so I CAN use sensorless homing which works however!! If I want to use e0/e1det WHILST I have sensorless homing enabled it causes issues because the skr3 pins file automatically reassigns the e0/e1det pins as x/y max endstops if sensorless homing is enabled, even if in the main marlin config.h x/y max endstops are commented out and that causes issues for using the e0/e1det pins WHEN using sensorless homing. 
[image]

radek8, 2022-07-11:
The pin file is fine.
If you define X_MIN stop in configuration.h and enable sensorless guidance, X_MAX is not defined.

radek8, 2022-07-11:
you should set

//#define X_DUAL_ENDSTOPS   
//#define Y_DUAL_ENDSTOPS   
#define X_HOME_DIR -1
#define Y_HOME_DIR -1

Tony-g80, 2022-07-11:
if you look at the screenshot for skr3 pins in last comment of mine you will see that if stall sensitivity is declared then it changes the X/ymax  to the e0/e1 pins even though the xmax are commented out in main config and min endstops are enabled as should be even for sensorless homing but if I try to use the e0/e1 they won’t work reliably

Tony-g80, 2022-07-11:
Yes that is what I have it set to

Tony-g80, 2022-07-12:
Already gave that one a try and marlin had a multi message failure and non compile when I took those out of the equation