bigtreetech/SKR-3 #5: Ender 3v2 DWIN rotary knob display support

WebDesignerlol, 2022-07-07:
Hello, I am quite versed in the ability to make my ender 3v2 rotary knob LCD work w both the SKR 1.4 turbo and the SKR 2 which are functionally similar to this board.

I can't figure out why this board doesn't seem to light up the screen like the other two did, and it was functional with the other two.

Any tips would be great. Did the pin tracing and lined up with diagrams and made my own custom cable and all that. Tx/Rx is going to the TFT connector hookup. There is a MISO and MOSI outputs on the EXP1 header and I am debating if I should be instead utilizing those instead of the two TFT pins as in previous board iterations.

Thanks for any info.

WebDesignerlol, 2022-07-08:
I have solved it.
I set LCD_SERIAL PORT 1 and SERIAL_PORT 2, along with the custom cable and TFT connection I wrote about above.

closed 2022-07-08 by WebDesignerlol

Sergeykurm, 2022-09-29:
Good afternoon! Are you doing well? Can you draw in detail how to connect the wires on the SCR 3 EZ board for the Ender 3 V2. screen? And what parameters to prescribe in Marlin.? I will be very grateful to you.