bigtreetech/SKR-3 #8: ESP32 board firmware not installing on SKR 3 EZ

dstrohl, 2022-07-24:
I seem to be unable to get the ESP32 wifi board to install on the system.

My steps:
I installed a ESP32 board on a BTT SKR 3 EZ motherboard per the docs, 
I configured Marlin serial port 3 per the docs and installed that version on the board.
I downloaded the current firmware from the wiki link and compiled it in Platform IO. *no changes in any config files.
I copied the firmware.bin file from the esp8266 directory to an sd card and named it esp3d.bin, then inserted that in the board's SD card slot and reset the board.
the red status lite blinked for a few minutes on both the SKR and ESP boards... then the status lite went out.
according to the docs, the file should be renamed to esp3d.cur afterwards, but the .bin file remained.

At this point. I am not seeing a wifi network with the name of ESP3D.  I am seeing one called ESP_xxxxxx (with the xxxxxx being the last 6 of the MAC address on the board)  when I connect to that network, (no password), it provides an address in the 192.168.4.x range, and the .1 host does not accept any connections.

Am I missing something? do I need to change something in the firmware?  is there a log file or other error message that would tell me what is going on that the firmware did not seem to install correctly?

The instructions on luc-github seem to imply that I need to install the firmware before installing the card on the board... but the instructions in the docs do not mention that.

PalosX, 2022-07-24:
don't change the firmware name. It doesn't recognize it, use the name it is made with firmare.bin. Did the same thing took forever to figure out

dstrohl, 2022-07-25:
I tried that and the motherboard board did not boot at all, the status light kept blinking after about 20 minutes I reset it.  I was assuming that the motherboard was then trying to load firmware.bin as it's own firmware.  not true?

PalosX, 2022-07-25:
Insert sd card with boartd off then power up. What a minute then check if file extension changed to CUR.

dstrohl, 2022-07-25:
yup, that's what I did, inserted the sd card into the skr slot with the power off, when the firmware was named firmware.bin, the power came up, and the status light never stopped blinking.. but the firmware.bin file was renamed to firmware.cur.  if I rebooted, the board never actually finished booting at that point.  If I put the old firmware.bin back on  (with marlin), it booted and copied the firmware and worked fine (but still no wifi).  (hence my thinking that what was happening was that the skr board was taking any file that was called firmware.bin and trying to load it into it's own memory, not passing it through to the esp board.

PalosX, 2022-07-25:
You have to use a boot loader watch these to videos
https://www.youtube.com/watch?v=NGgzw-XayEo
https://www.youtube.com/watch?v=wpZq2_nD0Ho

mrwakuhn, 2024-12-17:
I had the same issue (SKR3EZ bootloader not flashing the ESP WiFi module) and was able to use an Arduino to flash the ESP12S using the method in https://www.youtube.com/watch?v=3J1Ys7q8DTg