bigtreetech/SKR-3 #9: Unable to move stepper motors from Klipper

slavanyx, 2022-07-29:
Hi
I am getting 
**Unable to write tmc spi 'stepper_x' register GLOBALSCALER**
error when I try to home steppers on SKR 3 EZ, using EZ5160 drivers

below is my printer.cfg config, pretty standard with minimal changes. Everything boots up, can heat extruder/bed, when trying to home steppers getting this error. Can be related to software spi but since the card is pretty new not much examples on the web

Thank you!

# This file contains common pin mappings for the BigTreeTech SKR 3.
# To use this config, the firmware should be compiled for the
# STM32H743 with a "128KiB bootloader".

# See docs/Config_Reference.md for a description of parameters.

[include mainsail.cfg]

##### VzBoT############
#######################

[printer]
kinematics: corexy
max_velocity: 2000
max_accel: 50000
#max_accel_to_decel: 100000
max_z_velocity: 30
max_z_accel: 1500
square_corner_velocity: 20

########################
########################

#[mcu rpi]
#serial: /tmp/klipper_host_mcu

[mcu]
serial: /dev/serial/by-id/usb-Klipper_stm32h743xx_1C002C001651303232383230-if00

#[adxl345]
#cs_pin: rpi:None

#[resonance_tester]
#accel_chip: adxl345
#probe_points:
#    155,155,20  # an example

########################
########################
#[static_digital_output disable_max31865] pins: !PF8

[stepper_x]
step_pin: PD4
dir_pin: PD3
enable_pin: !PD6
microsteps: 16
rotation_distance: 40
endstop_pin: ^PC1
position_endstop: 0
position_max: 200
homing_speed: 50

[stepper_y]
step_pin: PA15
dir_pin: !PA8
enable_pin: !PD1
microsteps: 16
rotation_distance: 40
endstop_pin: ^PC3
position_endstop: 0
position_max: 200
homing_speed: 50

[stepper_z]
step_pin: PE2
dir_pin: PE3
enable_pin: !PE0
microsteps: 16
rotation_distance: 40
endstop_pin: ^PC0
position_endstop: 0.5
position_max: 200

[extruder]
step_pin: PD15
dir_pin: PD14
enable_pin: !PC7
microsteps: 16
nozzle_diameter: 0.400
filament_diameter: 1.750
heater_pin: PB3
sensor_type: EPCOS 100K B57560G104F
sensor_pin: PA2
rotation_distance: 5.57
full_steps_per_rotation: 200
filament_diameter: 1.750
max_extrude_only_velocity: 120
max_extrude_only_accel: 800
pressure_advance_smooth_time: 0.02
min_extrude_temp: 170
#control: pid
#pid_Kp: 22.2
#pid_Ki: 1.08
#pid_Kd: 114
min_temp: 0
max_temp: 250

[firmware_retraction]
retract_speed: 120
unretract_extra_length: 0
unretract_speed: 120

#[extruder1]
#step_pin: PD11
#dir_pin: PD10
#enable_pin: !PD13
#heater_pin: PB4
#sensor_pin: PA3
#...

[heater_bed]
heater_pin: PD7
sensor_type: Generic 3950
sensor_pin: PA1
#control: watermark
min_temp: -100
max_temp: 130

[fan]
pin: PB7

[heater_fan fan1]
pin: PB6

#[heater_fan fan2]
#pin: PB5

########################################
# EXP1 / EXP2 (display) pins
########################################

[board_pins]
aliases:
    # EXP1 header
    EXP1_1=PC5, EXP1_3=PB1, EXP1_5=PE9,  EXP1_7=PE11, EXP1_9=<GND>,
    EXP1_2=PB0, EXP1_4=PE8, EXP1_6=PE10, EXP1_8=PE12, EXP1_10=<5V>,
    # EXP2 header
    EXP2_1=PA6, EXP2_3=PE7, EXP2_5=PB2, EXP2_7=PC4,   EXP2_9=<GND>,
    EXP2_2=PA5, EXP2_4=PA4, EXP2_6=PA7, EXP2_8=<RST>, EXP2_10=<NC>

# See the sample-lcd.cfg file for definitions of common LCD displays.

########################################
# TMC2209 configuration
########################################

#[tmc2209 stepper_x]
#uart_pin: PD5
#run_current: 0.800
#diag_pin:

#[tmc2209 stepper_y]
#uart_pin: PD0
#run_current: 0.800
#diag_pin:

[tmc2209 stepper_z]
uart_pin: PE1
run_current: 0.800
diag_pin:

[tmc2209 extruder]
uart_pin: PC6
run_current: 0.600
diag_pin:

#[tmc2209 extruder1]
#uart_pin: PD12
#run_current: 0.600
#diag_pin:

########################################
# TMC2130 configuration
########################################

[tmc5160 stepper_x]
cs_pin: PD5
spi_software_miso_pin: PE15
spi_software_mosi_pin: PE13
spi_software_sclk_pin: PE14
diag1_pin: PC1
run_current: 1.800
sense_resistor: 0.075
interpolate: false
stealthchop_threshold: 0

[tmc5160 stepper_y]
cs_pin: PD0
spi_software_miso_pin: PE15
spi_software_mosi_pin: PE13
spi_software_sclk_pin: PE14
run_current: 1.800
sense_resistor: 0.075
interpolate: false
stealthchop_threshold: 0
diag1_pin: PC3

#[tmc2130 stepper_z]
#cs_pin: PE1
#spi_software_miso_pin: PE15
#spi_software_mosi_pin: PE13
#spi_software_sclk_pin: PE14
#run_current: 0.650
#stealthchop_threshold: 999999
#diag1_pin: PC0

#[tmc2130 extruder]
#cs_pin: PC6
#spi_software_miso_pin: PE15
#spi_software_mosi_pin: PE13
#spi_software_sclk_pin: PE14
#run_current: 0.800
#stealthchop_threshold: 999999
#diag1_pin: PC2

#[tmc2130 extruder1]
#cs_pin: PD12
#spi_software_miso_pin: PE15
#spi_software_mosi_pin: PE13
#spi_software_sclk_pin: PE14
#run_current: 0.800
#stealthchop_threshold: 999999
#diag1_pin: PA0

#*# <---------------------- SAVE_CONFIG ---------------------->
#*# DO NOT EDIT THIS BLOCK OR BELOW. The contents are auto-generated.
#*#
#*# [extruder]
#*# control = pid
#*# pid_kp = 18.160
#*# pid_ki = 1.131
#*# pid_kd = 72.869
#*#
#*# [heater_bed]
#*# control = pid
#*# pid_kp = 35.229
#*# pid_ki = 0.536
#*# pid_kd = 578.632

ninjadudexp, 2022-08-26:
having this issue as well.

AlexWild74, 2024-05-01:
check your TMCs, if one is broken you will have that globalscaler