bigtreetech/SKR-3 #10: No E stepper plug left for Z2! when enable dual z.

paintedrealms, 2022-07-31:
suspect that it needs a setting placed in the pins file so we can use ZBM.

In file included from c:\scratch\marlin-bugfix-2.0.9.3.x-skr3\marlin\src\pins\pins.h:876,
                 from c:\scratch\marlin-bugfix-2.0.9.3.x-skr3\marlin\src\inc\marlinconfig.h:35,
                 from buildroot/share/PlatformIO/scripts/common-dependencies.h:29:
c:\scratch\marlin-bugfix-2.0.9.3.x-skr3\marlin\src\pins\pins_postprocess.h:790:8: error: #error "No E stepper plug left for Z2!"
  790 |       #error "No E stepper plug left for Z2!"
      |        ^~~~~
CalledProcessError: Command '"C:\Users\Stephen\.platformio\packages\toolchain-gccarmnoneeabi\bin\arm-none-eabi-g++.exe" -D__MARLIN_FIRMWARE__ -DNDEBUG -DHAL_STM32 -DUSBCON -DUSBD_USE_CDC -DTIM_IRQ_PRIO=13 -DADC_RESOLUTION=12 -DPIN_SERIAL1_RX=PA_10 -DPIN_SERIAL1_TX=PA_9 -DPIN_SERIAL3_RX=PD_9 -DPIN_SERIAL3_TX=PD_8 -DPIN_SERIAL4_RX=PA_1 -DPIN_SERIAL4_TX=PA_0 -DSERIAL_RX_BUFFER_SIZE=1024 -DSERIAL_TX_BUFFER_SIZE=1024 -DTIMER_SERVO=TIM5 -DTIMER_TONE=TIM2 -DSTEP_TIMER_IRQ_PRIO=0 -DD_CACHE_DISABLED -D__MARLIN_DEPS__ -w -dM -E -x c++ buildroot/share/PlatformIO/scripts/common-dependencies.h' returned non-zero exit status 1.:
  File "C:\Users\Stephen\.platformio\penv\lib\site-packages\platformio\builder\main.py", line 186:
    env.SConscript(item, exports="env")
  File "C:\Users\Stephen\.platformio\packages\tool-scons\scons-local-4.3.0\SCons\Script\SConscript.py", line 597:
    return _SConscript(self.fs, *files, **subst_kw)
  File "C:\Users\Stephen\.platformio\packages\tool-scons\scons-local-4.3.0\SCons\Script\SConscript.py", line 285:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "C:\scratch\Marlin-bugfix-2.0.9.3.x-skr3\buildroot\share\PlatformIO\scripts\common-dependencies.py", line 247:
    apply_features_config()
  File "C:\scratch\Marlin-bugfix-2.0.9.3.x-skr3\buildroot\share\PlatformIO\scripts\common-dependencies.py", line 133:
    if not env.MarlinFeatureIsEnabled(feature):
  File "C:\Users\Stephen\.platformio\packages\tool-scons\scons-local-4.3.0\SCons\Util.py", line 742:
    return self.method(*nargs, **kwargs)
  File "C:\scratch\Marlin-bugfix-2.0.9.3.x-skr3\buildroot\share\PlatformIO\scripts\common-dependencies.py", line 216:
    load_marlin_features()
  File "C:\scratch\Marlin-bugfix-2.0.9.3.x-skr3\buildroot\share\PlatformIO\scripts\common-dependencies.py", line 204:
    define_list = run_preprocessor(env)
  File "C:\scratch\Marlin-bugfix-2.0.9.3.x-skr3\buildroot\share\PlatformIO\scripts\preprocessor.py", line 42:
    define_list = subprocess.check_output(cmd, shell=True).splitlines()
  File "C:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.8_3.8.2800.0_x64__qbz5n2kfra8p0\lib\subprocess.py", line 415:
    return run(*popenargs, stdout=PIPE, timeout=timeout, check=True,
  File "C:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.8_3.8.2800.0_x64__qbz5n2kfra8p0\lib\subprocess.py", line 516:
    raise CalledProcessError(retcode, process.args,

paintedrealms, 2022-07-31:
These are the files with changes/
[Configuration.h.txt](https://github.com/bigtreetech/SKR-3/files/9229799/Configuration.h.txt)
[Configuration_adv.h.txt](https://github.com/bigtreetech/SKR-3/files/9229800/Configuration_adv.h.txt)

xhejtman, 2022-08-04:
Set extruders 1 if you want to use Z2 instead of E1 (Configuration.h)

paintedrealms, 2022-08-06:
Changed extruders to 1: new output

In file included from Marlin\src\HAL\STM32\../../inc/MarlinConfig.h:50,
                 from Marlin\src\HAL\STM32\HAL.cpp:30:
Marlin\src\HAL\STM32\../../inc/SanityCheck.h:948:6: error: #error "BABYSTEP_HOTEND_Z_OFFSET requires 2 or more HOTENDS."
  948 |     #error "BABYSTEP_HOTEND_Z_OFFSET requires 2 or more HOTENDS."
      |      ^~~~~
Marlin\src\HAL\STM32\../../inc/SanityCheck.h:972:6: error: #error "NUM_RUNOUT_SENSORS must be either 1 or number of E steppers."
  972 |     #error "NUM_RUNOUT_SENSORS must be either 1 or number of E steppers."
      |      ^~~~~
Compiling .pio\build\STM32H743Vx_btt\src\src\HAL\STM32\tft\tft_ltdc.cpp.o
In file included from Marlin\src\HAL\STM32\../../inc/MarlinConfig.h:50,
                 from Marlin\src\HAL\STM32\HAL_SPI.cpp:27:
Marlin\src\HAL\STM32\../../inc/SanityCheck.h:948:6: error: #error "BABYSTEP_HOTEND_Z_OFFSET requires 2 or more HOTENDS."
  948 |     #error "BABYSTEP_HOTEND_Z_OFFSET requires 2 or more HOTENDS."
      |      ^~~~~
Marlin\src\HAL\STM32\../../inc/SanityCheck.h:972:6: error: #error "NUM_RUNOUT_SENSORS must be either 1 or number of E steppers."
  972 |     #error "NUM_RUNOUT_SENSORS must be either 1 or number of E steppers."
      |      ^~~~~
In file included from Marlin\src\HAL\STM32\../../inc/MarlinConfig.h:50,
                 from Marlin\src\HAL\STM32\MarlinSerial.cpp:27:
Marlin\src\HAL\STM32\../../inc/SanityCheck.h:948:6: error: #error "BABYSTEP_HOTEND_Z_OFFSET requires 2 or more HOTENDS."
  948 |     #error "BABYSTEP_HOTEND_Z_OFFSET requires 2 or more HOTENDS."
      |      ^~~~~
Marlin\src\HAL\STM32\../../inc/SanityCheck.h:972:6: error: #error "NUM_RUNOUT_SENSORS must be either 1 or number of E steppers."
  972 |     #error "NUM_RUNOUT_SENSORS must be either 1 or number of E steppers."
      |      ^~~~~
In file included from Marlin\src\HAL\STM32\../../inc/MarlinConfig.h:50,
                 from Marlin\src\HAL\STM32\Servo.cpp:27:
Marlin\src\HAL\STM32\../../inc/SanityCheck.h:948:6: error: #error "BABYSTEP_HOTEND_Z_OFFSET requires 2 or more HOTENDS."
  948 |     #error "BABYSTEP_HOTEND_Z_OFFSET requires 2 or more HOTENDS."
      |      ^~~~~
Marlin\src\HAL\STM32\../../inc/SanityCheck.h:972:6: error: #error "NUM_RUNOUT_SENSORS must be either 1 or number of E steppers."
  972 |     #error "NUM_RUNOUT_SENSORS must be either 1 or number of E steppers."
      |      ^~~~~
In file included from Marlin\src\HAL\STM32\../../inc/MarlinConfig.h:50,
                 from Marlin\src\HAL\STM32\eeprom_flash.cpp:27:
Marlin\src\HAL\STM32\../../inc/SanityCheck.h:948:6: error: #error "BABYSTEP_HOTEND_Z_OFFSET requires 2 or more HOTENDS."
  948 |     #error "BABYSTEP_HOTEND_Z_OFFSET requires 2 or more HOTENDS."
      |      ^~~~~
Marlin\src\HAL\STM32\../../inc/SanityCheck.h:972:6: error: #error "NUM_RUNOUT_SENSORS must be either 1 or number of E steppers."
  972 |     #error "NUM_RUNOUT_SENSORS must be either 1 or number of E steppers."
      |      ^~~~~
In file included from Marlin\src\HAL\STM32\../../inc/MarlinConfig.h:50,
                 from Marlin\src\HAL\STM32\eeprom_bl24cxx.cpp:32:
Marlin\src\HAL\STM32\../../inc/SanityCheck.h:948:6: error: #error "BABYSTEP_HOTEND_Z_OFFSET requires 2 or more HOTENDS."
  948 |     #error "BABYSTEP_HOTEND_Z_OFFSET requires 2 or more HOTENDS."
      |      ^~~~~
Marlin\src\HAL\STM32\../../inc/SanityCheck.h:972:6: error: #error "NUM_RUNOUT_SENSORS must be either 1 or number of E steppers."
  972 |     #error "NUM_RUNOUT_SENSORS must be either 1 or number of E steppers."
      |      ^~~~~
In file included from Marlin\src\HAL\STM32\../../inc/MarlinConfig.h:50,
                 from Marlin\src\HAL\STM32\eeprom_sram.cpp:27:
Marlin\src\HAL\STM32\../../inc/SanityCheck.h:948:6: error: #error "BABYSTEP_HOTEND_Z_OFFSET requires 2 or more HOTENDS."
  948 |     #error "BABYSTEP_HOTEND_Z_OFFSET requires 2 or more HOTENDS."
      |      ^~~~~
Marlin\src\HAL\STM32\../../inc/SanityCheck.h:972:6: error: #error "NUM_RUNOUT_SENSORS must be either 1 or number of E steppers."
  972 |     #error "NUM_RUNOUT_SENSORS must be either 1 or number of E steppers."
      |      ^~~~~
In file included from Marlin\src\HAL\STM32\../../inc/MarlinConfig.h:50,
                 from Marlin\src\HAL\STM32\eeprom_if_iic.cpp:32:
Marlin\src\HAL\STM32\../../inc/SanityCheck.h:948:6: error: #error "BABYSTEP_HOTEND_Z_OFFSET requires 2 or more HOTENDS."
  948 |     #error "BABYSTEP_HOTEND_Z_OFFSET requires 2 or more HOTENDS."
      |      ^~~~~
Marlin\src\HAL\STM32\../../inc/SanityCheck.h:972:6: error: #error "NUM_RUNOUT_SENSORS must be either 1 or number of E steppers."
  972 |     #error "NUM_RUNOUT_SENSORS must be either 1 or number of E steppers."
      |      ^~~~~
In file included from Marlin\src\HAL\STM32\../../inc/MarlinConfig.h:50,
                 from Marlin\src\HAL\STM32\eeprom_sdcard.cpp:31:
Marlin\src\HAL\STM32\../../inc/SanityCheck.h:948:6: error: #error "BABYSTEP_HOTEND_Z_OFFSET requires 2 or more HOTENDS."
[... 52 lines cut ...]
*** [.pio\build\STM32H743Vx_btt\src\src\HAL\STM32\eeprom_if_iic.cpp.o] Error 1
In file included from Marlin\src\HAL\STM32\tft\../../../inc/MarlinConfig.h:50,
                 from Marlin\src\HAL\STM32\tft\tft_fsmc.cpp:27:
Marlin\src\HAL\STM32\tft\../../../inc/SanityCheck.h:948:6: error: #error "BABYSTEP_HOTEND_Z_OFFSET requires 2 or more HOTENDS."
  948 |     #error "BABYSTEP_HOTEND_Z_OFFSET requires 2 or more HOTENDS."
      |      ^~~~~
[... cut ...]
Marlin\src\HAL\STM32\tft\../../../inc/SanityCheck.h:972:6: error: #error "NUM_RUNOUT_SENSORS must be either 1 or number of E steppers."
  972 |     #error "NUM_RUNOUT_SENSORS must be either 1 or number of E steppers."
      |      ^~~~~
*** [.pio\build\STM32H743Vx_btt\src\src\HAL\STM32\sdio.cpp.o] Error 1
*** [.pio\build\STM32H743Vx_btt\src\src\HAL\STM32\fastio.cpp.o] Error 1
*** [.pio\build\STM32H743Vx_btt\src\src\HAL\STM32\tft\gt911.cpp.o] Error 1
*** [.pio\build\STM32H743Vx_btt\src\src\HAL\STM32\tft\tft_fsmc.cpp.o] Error 1
*** [.pio\build\STM32H743Vx_btt\src\src\HAL\STM32\tft\tft_ltdc.cpp.o] Error 1

xhejtman, 2022-08-06:
Clearly obvious. 

Comment out: BABYSTEP_HOTEND_Z_OFFSET
Set NUM_RUNOUT_SENSORS to 1 or comment out.

paintedrealms, 2022-08-06:
Thanks.    The settings you put work.  However, it is interesting.  I want to eventually enable the last stepper on the board for dual extrusion since I have one more output.  Hopefully This is possible.

closed 2022-08-06 by paintedrealms

xhejtman, 2022-08-08:
This is misunderstanding, you cannot have independent Z axis AND dual extrusion because you have only 5 drivers (x,y,z,e0,e1). Those Z outputs are shared via single driver, so in case of independent Z axis, you plug second Z into E1 output. So you need to choose, what do you prefer.