bigtreetech/SKR-3 #11: powering raspberry pi over skr 3 ez

odbee, 2022-08-01:
Hello, it is advertised in the manual and on the website that the skr ez supports powering the raspberry pi directly. I don't see any explanation on this anywhere. I would really appreciate some help. thank you.

hanzov69, 2022-08-05:
I believe you may be mistaken- looking through the manual and the website, I see no mention of this. 

That said, there are several 5v rails available on the board, you could theoretically use one of these to power a Pi. Personally, I would recommend against it though.

iJebus, 2022-08-06:
I don't have any skin in the game here, but taken from page five of the manual in this repo.

> The power chip adopts TPS5450-5A, which supports DC12/24V power input.
The output current of the chip is up to 5A, and the peak value can reach 6A,
which **perfectly supports the power supply of Raspberry Pi**.

HTMLdu, 2023-03-27:
I have powered my raspberry pi with no issues from the probe power but your mileage may vary.

RoCry, 2023-04-19:
> 

Did you connected to GPIO on raspberry pi? I did this, but there are some warning like 'Under-Voltage Detected' or 'Currently Throttled'