bigtreetech/SKR-3 #12: Marlin 2.1 compatibility

xhejtman, 2022-08-03:
Hello,

should this board be compatible with vanila Marlin 2.1? I use mostly the same config as for 2.0.9 from this git, but after flash, I get error:
error:eeprom datasize error.
and board does not boot. Is this known problem?

Bakorlam, 2022-08-11:
@xhejtman  Have you tried enabling these settings?-
// @section eeprom
/**
 * EEPROM
 * Persistent storage to preserve configurable settings across reboots.
 *   M500 - Store settings to EEPROM.
 *   M501 - Read settings from EEPROM. (i.e., Throw away unsaved changes)
 *   M502 - Revert settings to "factory" defaults. (Follow with M500 to init the EEPROM.)
 */
#define EEPROM_SETTINGS     // Persistent storage with M500 and M501         <<<<<<< HERE
//#define DISABLE_M503        // Saves ~2700 bytes of flash. Disable for release!
#define EEPROM_CHITCHAT       // Give feedback on EEPROM commands. Disable to save PROGMEM.
#define EEPROM_BOOT_SILENT    // Keep M503 quiet and only give errors during first load
#if ENABLED(EEPROM_SETTINGS)
  #define EEPROM_AUTO_INIT  // Init EEPROM automatically on any errors.       <<<<<<< HERE
  #define EEPROM_INIT_NOW   // Init EEPROM on first boot after a new build.   <<<<<<< and HERE
#endif

xhejtman, 2022-08-14:
I have all except EEPROM_INIT_NOW. However, I moved to klipper.