bigtreetech/SKR-3 #13: SKR3 EZ + TMC2209

xhejtman, 2022-08-03:
Hello,

I have SKR3 EZ board with old TMC2209 drivers that should be compatible. However, I am getting:
```
Driver registers:                                                                                                                     
                X       0x00:00:00:00    Bad response!                                                                                
                Y       0xC0:10:00:00                                                                                                 
                Z       0x00:00:00:00    Bad response!                                                                                
                Z2      0x00:00:00:00    Bad response!                                                                                
                E       0x80:14:00:C0                                                                                                 
                                                                                                                                      
                                                                                                                                      
Testing X connection... Error: All LOW                                                                                                
Testing Y connection... OK                                                                                                            
Testing Z connection... Error: All LOW                                                                                                
Testing Z2 connection... Error: All LOW                                                                                               
Testing E connection... OK                                                                                                            
//action:notification TMC CONNECTION ERROR   
```

but despite of this, all axis can move, so it looks like a spurious error? Also it changes randomly, which connection is ok and which has an error. Z2 is connected to E1, number of extruders is set to 1.

Setix, 2022-08-08:
+1 getting the exact same issue. I tried 2 different boards, with all new EZ2209 drivers. Also present in Marlin 2.1

xhejtman, 2022-08-08:
@setix: did you have chance to try different drivers? I will try EZ5160 tomorrow.

Anyway, does Marlin 2.1 boots for you? I cannot boot it, it stuck after eeprom configuration crc mismatch.

Setix, 2022-08-08:
@xhejtman I have not tried any other drivers that EZ2209, I thought my board was bad, but after swapping it out I got the same results of "Bad Response" and "Error: All LOW" same as you. I was able to get Marlin 2.1 to boot, but it contains the same error on M122.

xhejtman, 2022-08-08:
@Setix so, could you share your configs for marlin 2.1?

Setix, 2022-08-08:
@xhejtman Here ya go. Obvs rename the extension back to h
[Configuration.txt](https://github.com/bigtreetech/SKR-3/files/9285307/Configuration.txt)
[Configuration_adv.txt](https://github.com/bigtreetech/SKR-3/files/9285308/Configuration_adv.txt)

xhejtman, 2022-08-09:
It does work with TMC5160 drivers, they are all connected.

Michellegson, 2022-08-11:
Getting a similar issue with the EZ2209. Except it's always the same X,Z,&E1 ports that report "all low". The X port will flicker between reporting "ok" and "all low". And I've also tried with 2 separate sets of SKR3EZ boards & EZ2209 drivers (even swapping the sets) as well - same exact issues. It's suspicious that the issue is always in every other driver port and persists regardless of TMC or EZ versions of the 2209s

xhejtman, 2022-08-11:
I also suspect that with CONNECTION error, driver is actually not set from default values as sound while axis moving is a bit strange/different to SKR 1.4 + TMC2209 drivers and the same motors.

Michellegson, 2022-08-11:
I've noticed the same, the strange noises is most prominent in my Z axis. Tho I haven't really tested out the E1 port since I don't have a dedicated motor for it. I also haven't tested movement in standalone mode to see if the noises were still there.

ninjadudexp, 2022-08-11:
firmware or compiler issue, i had this issue and someone else compiled a version for me. So big thanks to LostinTech from youtube.

he sent me the files and even when i compiled it, they still had the issues. But when he compiled the same one it worked just fine.

think the compilers are missing something.

but try thesee and get back to me.

(These are set for an ender but at least youll be able to see if you TMC goes away

just change to bin obviously.
[firmware.txt](https://github.com/bigtreetech/SKR-3/files/9311961/firmware.txt)
)

xhejtman, 2022-08-11:
So, do we know what is different?

```
PACKAGES: 
 - framework-arduinoststm32 @ 0.0.0 
 - framework-cmsis @ 2.50700.210515 (5.7.0) 
 - toolchain-gccarmnoneeabi @ 1.90201.191206 (9.2.1)
Converting Marlin.ino
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 14 compatible libraries
Scanning dependencies...
Dependency Graph
|-- TMCStepper @ 0.7.3
|   |-- SPI @ 1.0.0
|   |-- SoftwareSerial @ 1.0.0
|-- SlowSoftI2CMaster @ 1.0.0
|-- SlowSoftWire @ 0.0.0+20220803222019
|   |-- SlowSoftI2CMaster @ 1.0.0
|-- SPI @ 1.0.0
|-- EEPROM @ 2.0.1
|-- SoftwareSerial @ 1.0.0
|-- IWatchdog @ 1.0.0
|-- Wire @ 1.0.0
|-- Servo @ 1.1.2
```

ninjadudexp, 2022-08-11:
I could not figure out the difference I would need his vs code folder and
platformio libraries I think. I'm just guessing. I'll keep trying different
things later today. I exhausted myself checking pins and stuff to finally
find something that works.

Michellegson, 2022-08-12:
I'm putting this board on my Ender5. I tested the bin file and it did work for me, all ports reported back "ok"!! Do you have any of the code files, at least the config & config_adv?

> firmware or compiler issue, i had this issue and someone else compiled a version for me. So big thanks to LostinTech from youtube.
>
> he sent me the files and even when i compiled it, they still had [...]

ninjadudexp, 2022-08-12:
> I'm putting this board on my Ender5. I tested the bin file and it did work for me, all ports reported back "ok"!! Do you have any of the code files, at least the config & config_adv?

I do, but i suspect it might not help you. It should compile but it might revert to your old issue. Let me know if you compile and what the outcome was

I tried this myself and i could not get it to fix TCM when i compiled it. I have his whole Marlin folder and i still had issues, thats what leads me to believe its somehow a compiling issue.

[Configuration.txt](https://github.com/bigtreetech/SKR-3/files/9313509/Configuration.txt)
[Configuration_adv.txt](https://github.com/bigtreetech/SKR-3/files/9313511/Configuration_adv.txt)
.

xhejtman, 2022-08-12:
> I could not figure out the difference I would need his vs code folder and platformio libraries I think. I'm just guessing. I'll keep trying different things later today. I exhausted myself checking pins and stuff to finally find something that works.
> […](#)

isn't he able to capture build output?

Btw, did anyone run into issue that `variant_MARLIN_STM32H743VX.h` cannot be found as the repo contains `variant_MARLIN_STM32H743Vx.h` file (lowercase x).

xhejtman, 2022-08-12:
I think there might be a problem with UART/serial in general, as I use TFT35 display for printing and print freezes in some cases and it seems that it is serial communication between TFT and SKR that freezes because controll over USB works normal.

ninjadudexp, 2022-08-12:
I still have yet to try it. But I think you need to do a full uninstall of
vs code. Then reinstall everything.

xhejtman, 2022-08-13:
If working firmware is connected via USB cable to linux box, does it report something very similar?

`/dev/serial/by-id/usb-STMicroelectronics_MARLIN_STM32H743VX_CDC_in_FS_Mode_3761327C3330-if00` ?

ninjadudexp, 2022-08-13:
No idea. I gave up after testing a bunch of different theories. I switched
to klipper for now. Had no issues at all with the drivers. So it's a Marlin
issue I'm pretty sure.

JoaquinBerrios, 2022-08-16:
I'm getting this issue as well with an SKR 3 EZ and TMC2209EZ drivers. I have tried building very basic configurations that just enable the 5 stepper drivers with everything from the sources on this repo, to release, patched and nightly versions of 2.0 and 2.1 and always have the same issue of the printer reporting a TMC CONNECTION ERROR and "All LOW" responses to M122 commands that jump around from driver to driver. Sometimes just on a single driver, often on multiple. I have some BTT 2208 V3.0 old style drivers on order to see if they have the same issue.

I've also tried uninstalling/re-installing VSC and PIO, in Windows 10 and Linux (Ubuntu 22.04) to see if it's a build environment issue as hinted above, and all builds have the same issue. 

I'm able to start prints in this state, but the printer pauses and re-starts randomly when printing from Octoprint. I have not tried printing from a file on the SD card to see if the pauses still happen. 

Below are my configs for the latest build of 2.1.x nightly, where everything seems to be working, but I get the driver errors and the prints pause randomly. All on an Ender 3 Max with BL Touch, TFT35 E3, filament runout sensor, Mini UPS and dual Z motors.
[Configuration.txt](https://github.com/bigtreetech/SKR-3/files/9353465/Configuration.txt)
[Configuration_adv.txt](https://github.com/bigtreetech/SKR-3/files/9353466/Configuration_adv.txt)

xhejtman, 2022-08-16:
I moved to klipper and everything is ok. So perhaps you could try the same, as speaking about octoprint, you should have rpi or something..

JoaquinBerrios, 2022-08-16:
I would prefer to stay with Marlin since the TFT display is useless with Klipper, and I prefer to have the option to control the printer locally. I may load Klipper up just to confirm that there is no hardware issues, but I think I would prefer to stay with Marlin long term.

Setix, 2022-08-16:
You can stay with Marlin, in my experience even though M122 reports errors it doesnt seem to affect the actual performance of them. Mine still move as expected. You can also change to SPI drivers, those dont seem to have this issue.

xhejtman, 2022-08-16:
> You can stay with Marlin, in my experience even though M122 reports errors it doesnt seem to affect the actual performance of them. Mine still move as expected. You can also change to SPI drivers, those dont seem to have this issue.

I am pretty sure, it actually does not do any settings of TMC drivers, it is ok, if you do not need it though.

vwsurfer, 2022-08-17:
Hey everyone, Did anyone fix this issue? I have the SKR 3 with tmc2208 with the same issue.

ninjadudexp, 2022-08-17:
It's an issue with compiling Marlin. If you can find someone else to
compile it chances are it might work. I look through a ton of different
directories couldn't get it to work for myself.  But I can confirm it is
definitely not a hardware issue, as using klipper this issue doesn't
happen.

JoaquinBerrios, 2022-08-17:
I've tried building this in multiple machines on multiple OS (Windows 10, 11, Ubuntu 22.04) and have the issues in all cases. If there is some magic sauce to setting up VSC/PIO to get this to work, I would love to know it, but it's not as simple as "build on another box/OS".
Maybe the builds that are working are setting build flags that are not set in the default environment, or linking against non-default libraries?

ninjadudexp, 2022-08-17:
I've taken full folders and libraries of others who compiled a working
version for me. And it does not work, same exact files and folders and
version of VB on the other works just fine. Not really sure where to
continue looking.

Bakorlam, 2022-08-17:
I've seen a bunch of mentions to the TMC lib provided by BTT, but not sure how to use it exactly...  [https://github.com/bigtreetech/TMCStepper](url)
Also saw in another comment to use that lib with -DHAVE_SW_SERIAL which I added to my env flags but didn't fix the issue, but like I said... I'm not even sure I was using it correctly. It also hasn't been updated since Oct 2019..

JoaquinBerrios, 2022-08-17:
I saw those threads as well somewhere, but I think I also remember reading that the BTT TMCStepper library breaks the build on newer releases of Marlin, which makes sense since it hasn't been update since 2019. I'm about two days away from putting the crappy stock board back in my printer and throwing this thing in the trash. Then maybe run Klipper on that. :-P

Bakorlam, 2022-08-17:
I got it to compile with that lib included with the build flag...but yeah no idea.. can confirm Skr3 EZ with EZ2209 works with Klipper and RepRap

vwsurfer, 2022-08-17:
If i make the basic config. board, ini, and tmc, port -1 i get no error.  but obviously thats no good.

JoaquinBerrios, 2022-08-18:
So if I build to only use the first four stepper drivers, I get no connection errors. If I try to enable the fifth driver as either Z2 or E1 I get the errors. I swapped the driver sticks around, and it makes no difference which sticks are where. Maybe bad board or pin definitions? I'll do more digging later, but thought I'd throw this out there to see it it leads anyone else somewhere...

ninjadudexp, 2022-08-18:
Could it be that you have all the drivers enabled by don't have anything
actually connected to it. Don't see how that would be an issue but sounds
odd that it works when you disable

JoaquinBerrios, 2022-08-18:
All the drivers have a motor connected. They all move even when the drivers are reporting the connection errors.

JoaquinBerrios, 2022-08-18:
I will add that the file Michellegson posted above that worked for some also only had only four steppers enabled.

vwsurfer, 2022-08-18:
Hello all, Last night i got someone to build a basic bin file for the skr3 with tmc 2208, i built the same config. his bin file gave me no tmc error. Mine did. Same config different pc. So he sent me the .h the .avd and the platform.ini . So if i copy those file to my marlin folder that should work? It doesnt. Is this problem VScode?

JoaquinBerrios, 2022-08-18:
vwsurfer, If that is the case, then it may be an VSCode or environment issue. Do you know what OS the person who did the good build is using? I've tried fresh VSC installs in new Ubuntu 22.04 and Windows 10/11 VMs and they all have the same issues. I would not have expected that builds using the same code in the same build tool end up with different binaries. Can you hash the binary you got and the one you built and see if they have the same hash? If one works and one doesn't I doubt it, but would be interesting to confirm.

JoaquinBerrios, 2022-08-18:
I also wanted to update that while the printer boots without errors with only four drivers enabled, they appear once I try to use the printer. Wondering if this is a more generic serial communication issue if it's a bit better when addressing less drivers, but still craps out once you start pushing traffic on the serial connections to the drivers...

vwsurfer, 2022-08-18:
JoaquinBerrios, I will be working on this untill its fixed or in the bin. I only know the basics of marlin.  I will get both bin files checked.  I tried vscode on my mac  and made the same config but got the errors.

Bakorlam, 2022-08-19:
Started a new firmware build, fresh Windows/VScode install, only 4 steppers enabled and even physically removed my 5th driver.... TMC CONNECTION ERROR....but on the bright side my tft works in touch mode now..

Pardiman, 2022-08-20:
Hello,

I'm new and don't know much, but I join in case I can help.

My problem is the same as everyone describes on the X, Z and Z2 axes. The X axis switches between reporting "good" and "all low", but always with strange values. In addition, the Z axis has no strength and loses steps easily.

I have tested the "firmware.txt" that ninjadudexp provided and gives me a correct M122, but not being able to change the configuration I am missing the Z2 axis and I can not connect the screen of my Artillery X1.

I have ordered right now five EZ5160s, which I was already thinking about buying. Being a clone of EZ2209 and cheap, I have also ordered some EZ6609 to try. When they arrive (15 days) I will try them, but I would like to solve the issue of the EZ2209.

Pardiman, 2022-08-21:
Reading what you say about environment problems, after some problems and a very long afternoon, I compiled on a Raspberry Pi 4 with Raspberry OS of 32bis and 64bits. 

I also compiled a firmware with PlatformIO Command-Line without VS Code on Windows 11.

But I always get the same "TMC CONNECTION ERROR" problem on the same axes as always.

JoaquinBerrios, 2022-08-21:
Pardiman, same here. I tried building on a raspberry pi 4 and still have the issues. I also tried updating the TMCStepper library with the new commits after v 0.7.3 since some seemed related to serial issue (although nothing specific to TMC2009 drivers), but still getting the errors. I also have an new board on order, but the fact that my current board seems to work with the firmware posted above by Michellegson kind of leads me to believe it is not a board error. This just seems really weird to me. At this point, I'm nearing the limits of my patience an will request an RMA from BTT. I'm tired of troubleshooting this, and they have not even responded to my official support request yet. I'm going to request an RMA, and if I don't get a response on that, I'll contest the credit card charge with my bank.

JoaquinBerrios, 2022-08-21:
Also, I know it's a long shot that anyone who has built working firmware is trolling git issues, but if you've been able to build Marlin for an SKR 3 or SKR 3 EZ with TMC stepper drivers, could you give some details on what your build environment is and how you built it? Bonus points for pasting in the build log from VSC showing the messages during that build.

Pardiman, 2022-08-21:
did contact Bigtreetech support through the BIQU website, they answered me this Friday asking me if I had tried to switch the drivers.

I replied by sending them the link to this thread, so that they could see that there are more people with the same problem and that it does not seem like hardware damage.

But I understand that this GitHub is theirs, so they should already know.

Today I don't think I'll do anything, I'm a little tired too.

Caz3mir, 2022-08-21:
Hello everyone. I am having the same issue. I have tried all Marlin versions including the one from here. I get the same TMC CONNECTION ERROR on my build just like [xhejtman](https://github.com/xhejtman). I have tried the firmware made by ListinTech and i have all OK. If anyone has a clue please post here.
Brand new SKR3EZ with 5 x TMC2209EZ modules that i am trying to fit to my Ender 5 Plus.
Cheers

FlyingSamson, 2022-08-21:
I have the same problem, but as @vwsurfer also on a SRK3 (not EZ) but with TMC2209. In my case M122 reports "Bad response" and "All low" for all 5 drivers. I tried building different versions of Marlin on a Ubuntu 20.04 machine using VS Code 1.70.2, PIO Core 6.1.4, PIO Home 3.4.3 and on a MacOS 10.13 machine using Atom, PIO Core 6.1.4, PIO Home 3.4.3.

Setix, 2022-08-21:
Has anyone tried compiling Marlin with the BTT EZ drivers found [Here]?(https://github.com/bigtreetech/BIGTREETECH-EZ-Driver) 

I would try this but I’m out of town at the moment

Caz3mir, 2022-08-21:
> I have the same problem, but as @vwsurfer also on a SRK3 (not EZ) but with TMC2209. In my case M122 reports "Bad response" and "All low" for all 5 drivers. I tried building different versions of [...]

How are you powering the board? All low could be because you are powering from usb. In my case it is getting 24v from a 4A power supply.

Setix, 2022-08-21:
> 

Im using a Meanwell UHP-500R-24 so it’s definitely not a lack of amperage.

Caz3mir, 2022-08-21:
> Has anyone tried compiling Marlin with the BTT EZ drivers found [Here](https://github.com/bigtreetech/BIGTREETECH-EZ-Driver) ?
> 
> I would try this but I’m out of town at the moment

There is no software there, only 3d files and pdfs. Besides, the TMCStepper library is loaded automatically by Platformio, i tried 0.7.3 and a few older ones. I was unable to load the 0.5.5 old one from here https://github.com/bigtreetech/TMCStepper

Setix, 2022-08-21:
> There is no software there, only 3d files and pdfs. Besides, the TMCStepper library is loaded automatically by Platformio, i tried 0.7.3 and a few older ones. I was unable to load the 0.5.5 old one [...]

I JUST realized that looking through the repo on my phone…ugh…ok disregard

FlyingSamson, 2022-08-21:
> How are you powering the board? All low could be because you are powering from usb. In my case it is getting 24v from a 4A power supply.

I used a Siglent lab power supply to generate 24V with Amperage limited to 3A (but that Amerage is not even close to being exhausted)

TOTA42, 2022-08-21:
I have the same issue where X, Z, and E1 giving Error: All low as everybody else is having. Running [ninjadudexp](https://github.com/ninjadudexp)'s firmware.bin showed the connection worked as well. I did a test by swapping the pins around in the pins_BTT_SKR_V3_0_common.h and it follows the pins (i.e. if I swap X & Y and Z & E, the errors are now at Y, E, and E1) so I'm wondering if somehow the IO pins get configured differently or there's some conflict issue. It would be great to know how that firmware.bin differs from everything else. I'm using this for my Ender 3 v2 and this is the only issue I have so far.

Pardiman, 2022-08-22:
Any problem with software serial? Can we use hardware serial on the SKR 3 EZ board?

At this point I have no knowledge of how to do it. I have come to see that "Serial1" (in pins_BTT_SKR_V3_0_common.h) is not defined, but I do not know if any other ports have to be sacrificed or if physical pins are necessary.

Bakorlam, 2022-08-22:
See something I've been wondering about too; from page 13 of the SKR3 EZ manual-

3.3.2 TMC/EZ-driven UART/SPI Mode
TMC series drivers do not support the use of both UART and SPI drivers at the same time, for example: X, Y-axis use TMC/EZ2209 (UART), Z, E0 axis use TMC/EZ5160 (SPI).
**_The EZ series drivers support the simultaneous use of both UART and SPI drivers._**

Pardiman, 2022-08-22:
[Bakorlam](https://github.com/Bakorlam), I think the manual refers to the fact that the EZ series, unlike normal TMCs, does not use physical pins for its configuration, does almost everything by software and will have designed or programmed something to make UART and SPI drivers compatible on the same board.

My problem is that the EZ2209 are exclusively UART by software, but if they do not communicate with the board to be able to configure them by software, they are useless. My Z axis has a very low torque and loses steps, but if I do not have communication I can not raise the current via software, or at least, I raise the current and does not improve the torque. I think that those of you who have normal TMC2209, at least, can regulate the current adjustment potentiometer, even if you lose the other characteristics.

By the way, with the firmware of [ninjadudexp](https://github.com/ninjadudexp) I have not managed to improve the torque of the Z axis either.

[Bakorlam](https://github.com/Bakorlam), [xhejtman](https://github.com/xhejtman) tested with TMC5160 drivers and it seems that SPI works correctly. I have ordered some EZ5160 drivers, to test, although they are totally oversized for what I needed.

> It does work with TMC5160 drivers, they are all connected.

Sorry if I've been a little confused in explaining it, I'm not sure it works exactly like that either.

JoaquinBerrios, 2022-08-22:
So I tried Klipper and everything works as expected, so definitely not a hardware issue. I'm able to debug and configure all the drivers without issue, they all work as expected and don't appear to be loosing steps. 

At this point I would lean towards an incompatibility of the TMCStepper library with other dependencies of this board. The 2209 and 5160 drivers are handled differently by the library, where the 2208 and 2209's share a ton of the code and both are having issues with this board. Unfortunately, I'm not a good enough coder to troubleshoot beyond that. 

It's still weird to me that some people are able to get a build that works while others can't using the same version of the library. It would be nice to have a build log from someone who built this successfully to compare (HINT HINT BTT!).

vwsurfer, 2022-08-22:
JoaquinBerrios My SKR 3 Is fixed with no errors. I Built the Marlin for my printer tried it and got the TMC error, then i sent the  .h and .adv and platform.ini to my friend. He compiled it in his vscode V1.60 and  sent me the bin file, Works perfectly with no errors. Its something to do with VScode or VScode enviroment.

[70 more replies]