bigtreetech/SKR-3 #15: Thermistor type 11 throws compile error

Setix, 2022-08-08:
Using `#define TEMP_SENSOR_BED 11` on line 508 of Configuration.h of the BTT version of Marlin-bugfix-2.0.9.x throws the following error on compile

`In file included from Marlin\src\inc/MarlinConfigPre.h:37,
                 from Marlin\src\inc/MarlinConfig.h:28,
                 from Marlin\src\MarlinCore.h:24,
                 from Marlin\src\MarlinCore.cpp:31:
Marlin\src\module/thermistor/thermistors.h:217:22: error: 'temptable_1' was not declared in this scope; did you mean 'temptable_11'?
  217 | #define _TT_NAME(_N) temptable_ ## _N
      |                      ^~~~~~~~~~
Marlin\src\inc/../core/macros.h:280:37: note: in definition of macro 'COUNT'
  280 | #define COUNT(a)            (sizeof(a)/sizeof(*a))
      |                                     ^
Marlin\src\module/thermistor/thermistors.h:218:21: note: in expansion of macro '_TT_NAME'
  218 | #define TT_NAME(_N) _TT_NAME(_N)
      |                     ^~~~~~~~
Marlin\src\module/thermistor/thermistors.h:229:23: note: in expansion of macro 'TT_NAME'
  229 |   #define TEMPTABLE_1 TT_NAME(TEMP_SENSOR_1)
      |                       ^~~~~~~
Marlin\src\module/thermistor/thermistors.h:230:33: note: in expansion of macro 'TEMPTABLE_1'
  230 |   #define TEMPTABLE_1_LEN COUNT(TEMPTABLE_1)
      |                                 ^~~~~~~~~~~
Marlin\src\module/thermistor/thermistors.h:327:46: note: in expansion of macro 'TEMPTABLE_1_LEN'
  327 | static_assert(255 > TEMPTABLE_0_LEN || 255 > TEMPTABLE_1_LEN || 255 > TEMPTABLE_2_LEN || 255 > TEMPTABLE_3_LEN
      |                                              ^~~~~~~~~~~~~~~
Marlin\src\module/thermistor/thermistors.h:217:22: error: 'temptable_1' was not declared in this scope; did you mean 'temptable_11'?
  217 | #define _TT_NAME(_N) temptable_ ## _N
      |                      ^~~~~~~~~~
Marlin\src\inc/../core/macros.h:280:48: note: in definition of macro 'COUNT'
  280 | #define COUNT(a)            (sizeof(a)/sizeof(*a))
      |                                                ^
Marlin\src\module/thermistor/thermistors.h:218:21: note: in expansion of macro '_TT_NAME'
  218 | #define TT_NAME(_N) _TT_NAME(_N)
      |                     ^~~~~~~~
Marlin\src\module/thermistor/thermistors.h:229:23: note: in expansion of macro 'TT_NAME'
  229 |   #define TEMPTABLE_1 TT_NAME(TEMP_SENSOR_1)
      |                       ^~~~~~~
Marlin\src\module/thermistor/thermistors.h:230:33: note: in expansion of macro 'TEMPTABLE_1'
  230 |   #define TEMPTABLE_1_LEN COUNT(TEMPTABLE_1)
      |                                 ^~~~~~~~~~~
Marlin\src\module/thermistor/thermistors.h:327:46: note: in expansion of macro 'TEMPTABLE_1_LEN'
  327 | static_assert(255 > TEMPTABLE_0_LEN || 255 > TEMPTABLE_1_LEN || 255 > TEMPTABLE_2_LEN || 255 > TEMPTABLE_3_LEN
      |                                              ^~~~~~~~~~~~~~~
*** [.pio\build\STM32H743Vx_btt\src\src\MarlinCore.cpp.o] Error 1
In file included from Marlin\src\core\../inc/MarlinConfigPre.h:37,
                 from Marlin\src\core\utility.h:24,
                 from Marlin\src\core\utility.cpp:23:
Marlin\src\core\../module/thermistor/thermistors.h:217:22: error: 'temptable_1' was not declared in this scope; did you mean 'temptable_11'?
  217 | #define _TT_NAME(_N) temptable_ ## _N
      |                      ^~~~~~~~~~
Marlin\src\core\../inc/../core/macros.h:280:37: note: in definition of macro 'COUNT'
  280 | #define COUNT(a)            (sizeof(a)/sizeof(*a))
      |                                     ^
Marlin\src\core\../module/thermistor/thermistors.h:218:21: note: in expansion of macro '_TT_NAME'
  218 | #define TT_NAME(_N) _TT_NAME(_N)
      |                     ^~~~~~~~
Marlin\src\core\../module/thermistor/thermistors.h:229:23: note: in expansion of macro 'TT_NAME'
  229 |   #define TEMPTABLE_1 TT_NAME(TEMP_SENSOR_1)
      |                       ^~~~~~~
Marlin\src\core\../module/thermistor/thermistors.h:230:33: note: in expansion of macro 'TEMPTABLE_1'
  230 |   #define TEMPTABLE_1_LEN COUNT(TEMPTABLE_1)
      |                                 ^~~~~~~~~~~
Marlin\src\core\../module/thermistor/thermistors.h:327:46: note: in expansion of macro 'TEMPTABLE_1_LEN'
  327 | static_assert(255 > TEMPTABLE_0_LEN || 255 > TEMPTABLE_1_LEN || 255 > TEMPTABLE_2_LEN || 255 > TEMPTABLE_3_LEN
      |                                              ^~~~~~~~~~~~~~~
Marlin\src\core\../module/thermistor/thermistors.h:217:22: error: 'temptable_1' was not declared in this scope; did you mean 'temptable_11'?
  217 | #define _TT_NAME(_N) temptable_ ## _N
      |                      ^~~~~~~~~~
Marlin\src\core\../inc/../core/macros.h:280:48: note: in definition of macro 'COUNT'
  280 | #define COUNT(a)            (sizeof(a)/sizeof(*a))
      |                                                ^
Marlin\src\core\../module/thermistor/thermistors.h:218:21: note: in expansion of macro '_TT_NAME'
  218 | #define TT_NAME(_N) _TT_NAME(_N)
      |                     ^~~~~~~~
Marlin\src\core\../module/thermistor/thermistors.h:229:23: note: in expansion of macro 'TT_NAME'
  229 |   #define TEMPTABLE_1 TT_NAME(TEMP_SENSOR_1)
      |                       ^~~~~~~
Marlin\src\core\../module/thermistor/thermistors.h:230:33: note: in expansion of macro 'TEMPTABLE_1'
  230 |   #define TEMPTABLE_1_LEN COUNT(TEMPTABLE_1)
      |                                 ^~~~~~~~~~~
Marlin\src\core\../module/thermistor/thermistors.h:327:46: note: in expansion of macro 'TEMPTABLE_1_LEN'
  327 | static_assert(255 > TEMPTABLE_0_LEN || 255 > TEMPTABLE_1_LEN || 255 > TEMPTABLE_2_LEN || 255 > TEMPTABLE_3_LEN
      |                                              ^~~~~~~~~~~~~~~
*** [.pio\build\STM32H743Vx_btt\src\src\core\utility.cpp.o] Error 1`

96Doma, 2023-01-30:
Hi, It works for me! Marlin 2.1.2.

Temerev-KS, 2023-07-12:
same problem

Temerev-KS, 2023-07-12:
@Setix, were you able to find a solution to this issue?