bigtreetech/SKR-3 #21: Cannot communicate with SKR 3 EZ over USB

sheski3, 2022-09-13:
Hi,  When I attach the SKR3 EZ to my computer with a USB cable, I cannot communicate using YAT (among others) as no COM port is created.  The indicator lights show that the RepRap firmware has been installed.  What am I doing wrong?

Jerry

JoaquinBerrios, 2022-09-19:
Check that the USB/CAN switch is not depressed. The switch is right behind the USB port. Should be up for USB mode and pressed in for CAN mode. If it's in CAN mode the USB port will be disabled.

sheski3, 2022-09-19:
I was able to connect, but I get the same response " Locked $" to a M115 command.

reviloo0, 2023-08-04:
I have the same problem how did you connect?

Type2Enthusiasm, 2025-02-03:
My USB/CAN switch is not depressed, but no COM port is created. Any ideas?