bigtreetech/SKR-3 #23: SKR3 tmc2209 marlin error

alenik88, 2022-09-19:
Why marlin send error when power on printer? I use tmc2209 v1.2 uart by bigtreetech. I can't move any axis. When compile firmware not have error but printer can't move axis. 

[image]

[image]

[image]

I not understand why this problem, with skr2 vb when put tmc2209 v1.2 all work fine withouth any problem....

Bakorlam, 2022-09-23:
Hey, take a look at issue #13 [https://github.com/bigtreetech/SKR-3/issues/13](url)

Varghesemela, 2022-10-07:
I am facing the same issue. I uploaded the firmware for XYZ axis. The board inits and the serial response: 
Testing X connection... Error: All LOW
Testing Y connection... Error: All LOW
Testing Z connection... Error: All LOW
The motors are not engaged and i feel the TMC is not communicating properly.

Bakorlam, 2022-10-07:
@Varghesemela have you taken a look at issue https://github.com/bigtreetech/SKR-3/issues/13 [https://github.com/bigtreetech/SKR-3/issues/13](https://github.com/bigtreetech/SKR-3/issues/url)

Varghesemela, 2022-10-11:
I went through the issue @Bakorlam thank you for mentioning it here.
I would also mention the fixes that I did which worked for me and I can safely say that everything is working fine now.

I have made some changes in 3 files: 'stm32h7.ini' file, marlin_STM32H743Vx.json file and platformio.ini file, those are as below:
Probably the first change is the crucial one, rest of these are just added changes, but this is what my config looks like.

1. stm32h7.ini

  Added a line above upload+protocol = cmsis-dap(line 39)
  `board_build.f_cpu = 480000000L`
  changed platform and platform-packages lines with the ones mentioned below
  ```
  platform                    = ststm32@>14.1.0
  platform_packages           = framework-arduinoststm32@4.20200.220530
                              toolchain-gccarmnoneeabi@>1.90201.191206
  ```
  
  With these two changes i could use the SKR3 with TMC in uart mode

2. marlin_STM32H743Vx.json
  Change fcpu to `"f_cpu": "480000000L"`

3. platformio.ini
 Changed default_envs to `default_envs = STM32H743Vx_btt`

Moebe1, 2022-10-14:
OMG, thank you for this. I was losing my mind trying to figure out why my TMC2226 drivers were throwing the TMC connection Error.

In my case, I was unable to get VS Code/PIO to download the correct arduinostm and toolchain version - I kept getting a path not found warning and it would be stuck in a loop looking for mirrors.

The cure was to head to C:\Users\{username}\.platformio and delete the entire folder > Relaunch VS Code and wait for PlatformIO to re-install. The build then succeeded.

`Could someone from BTT please update the Readme to capture this information? Use the info that @Varghesemela has so helpfully provided. For PlatformIO errors in VS Code not solved by cleaning the build environment, please delete the .platformio folder.`

I tried using the Firmware that BTT has - but that fails too since the stm32h7.ini file still contains the incorrect Platform_package references.

I came so close to returning the board - BTT please update your documentation.

alenik88, 2022-10-16:
in my cases, with this change now not have "tmc connection error" on display but no axis move in any directions...

Bakorlam, 2022-10-16:
@alenik88 I had that happen too after I tried the fix on a firmware I'd already been working on. I just downloaded a new zip of the same version of Marlin and copied the settings over. Worked just fine after that!