bigtreetech/SKR-3 #24: BTT TFT35 E3 V3.0.1 "no printer attached"

amad3v, 2022-09-19:
Hi,
The **TFT35 E3 v3.0.1** connected to **SKR3-EZ** shows **"no printer attached"**

### SKR3-EZ
Using the latest marlin firmware with the following changes:
- file: Configuration.h
```cpp
#define STRING_CONFIG_H_AUTHOR "(18.9.22, SKR3EZ)" // Who made the changes.
#define MOTHERBOARD BOARD_BTT_SKR_V3_0_EZ
#define SERIAL_PORT -1 // Enable TFT serial port
#define BAUDRATE 115200
#define SERIAL_PORT_2 1 // Enable USB emulated serial port
#define X_DRIVER_TYPE  TMC2209
#define Y_DRIVER_TYPE  TMC2209
#define Z_DRIVER_TYPE  TMC2209
#define Z2_DRIVER_TYPE TMC2209
#define E0_DRIVER_TYPE TMC2209
#define EEPROM_SETTINGS     // Persistent storage with M500 and M501
#define SDSUPPORT
#define REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER
```
- file: Configuration_adv.h
```cpp
#define LONG_FILENAME_HOST_SUPPORT   
#define AUTO_REPORT_SD_STATUS         // Auto-report media status with 'M27 S<seconds>'
#define SDCARD_CONNECTION ONBOARD
#define BABYSTEPPING
#define EMERGENCY_PARSER
#define SERIAL_FLOAT_PRECISION 4
#define SENSORLESS_HOMING // StallGuard capable drivers only
#define IMPROVE_HOMING_RELIABILITY
#define TMC_DEBUG
#define AUTO_REPORT_POSITION
#define M115_GEOMETRY_REPORT
#define M114_DETAIL         // Use 'M114` for details to check planner calculations
#define REPORT_FAN_CHANGE   // Report the new fan speed when changed by M106 (and others)
#define HOST_ACTION_COMMANDS
#define HOST_PROMPT_SUPPORT           // Initiate host prompts to get user feedback
#define HOST_STATUS_NOTIFICATIONS   // Send some status messages to the host as notifications
```
### TFT35 E3
Firmware used: [BIGTREE_TFT35_V3.0_E3.27.x.bin](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/blob/master/Copy%20to%20SD%20Card%20root%20directory%20to%20update/BIGTREE_TFT35_V3.0_E3.27.x.bin) with the default `config.ini` file, yet the info screen shows `GD_TFT35_E3_V3.0`

[image]

[image]

### Output
the output of the `M115` command is
```
SENDING:M115
FIRMWARE_NAME:Marlin bugfix-2.1.x (Sep 19 2022 09:48:14) SOURCE_CODE_URL:github.com/MarlinFirmware/Marlin PROTOCOL_VERSION:1.0 MACHINE_TYPE:jm3d EXTRUDER_COUNT:1 UUID:cede2a2f-41a2-4748-9b12-c55c62f367ff
Cap:SERIAL_XON_XOFF:0
Cap:BINARY_FILE_TRANSFER:0
Cap:EEPROM:1
Cap:VOLUMETRIC:1
Cap:AUTOREPORT_POS:1
Cap:AUTOREPORT_TEMP:1
Cap:PROGRESS:0
Cap:PRINT_JOB:1
Cap:AUTOLEVEL:0
Cap:RUNOUT:0
Cap:Z_PROBE:0
Cap:LEVELING_DATA:0
Cap:BUILD_PERCENT:0
Cap:SOFTWARE_POWER:0
Cap:TOGGLE_LIGHTS:0
Cap:CASE_LIGHT_BRIGHTNESS:0
Cap:EMERGENCY_PARSER:1
Cap:HOST_ACTION_COMMANDS:1
Cap:PROMPT_SUPPORT:1
Cap:SDCARD:1
Cap:MULTI_VOLUME:0
Cap:REPEAT:0
Cap:SD_WRITE:1
Cap:AUTOREPORT_SD_STATUS:1
Cap:LONG_FILENAME:1
Cap:LFN_WRITE:0
Cap:CUSTOM_FIRMWARE_UPLOAD:0
Cap:EXTENDED_M20:1
Cap:THERMAL_PROTECTION:1
Cap:MOTION_MODES:0
Cap:ARCS:1
Cap:BABYSTEPPING:1
Cap:CHAMBER_TEMPERATURE:0
Cap:COOLER_TEMPERATURE:0
Cap:MEATPACK:0
Cap:CONFIG_EXPORT:0
area:{full:{min:{x:0.0000,y:0.0000,z:0.0000},max:{x:200.0000,y:200.0000,z:200.0000}},work:{min:{x:0.0000,y:0.0000,z:0.0000},max:{x:200.0000,y:200.0000,z:200.0000}}}

```

amad3v, 2022-09-19:
I just found out that the screen is not powered by an **STM32F207VCT6** but by **GD32F205VCT6**, which explains having `GD_TFT35_E3_V3.0` in the info screen, so I uploaded the [BIGTREE_GD_TFT35_V3.0_E3.27.x.bin](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/blob/master/Copy%20to%20SD%20Card%20root%20directory%20to%20update/BIGTREE_GD_TFT35_V3.0_E3.27.x.bin) but it didn't change anything.

I switched to Marlin Mode and I got an error!! 

[image]

I pressed reset and restarted with no luck...

radek8, 2022-09-19:
Send a photo of the cable plugged in to TFT

amad3v, 2022-09-19:
[image]

[image]

radek8, 2022-09-20:
Do you have a properly oriented serial cable? 
In the second photo you can not see the color indexes on the cable. 
Check the conductives of the serial (black) cable.

amad3v, 2022-09-20:
I checked the wires connectivity and it's fine.

**Main board connection**
[image] [image]

**Screen connector**
[image] [image]

radek8, 2022-09-20:
Send a photo of the cable marking on the side of the display

amad3v, 2022-09-20:
The last 2 pictures are from the display side!!

radek8, 2022-09-20:
OK, the wiring looks fine.

radek8, 2022-09-20:
Do you have the same serial port communication speed set on the display?

radek8, 2022-09-20:
Does the printer respond to the M503 command?

amad3v, 2022-09-20:
The printer works fine with pronterface.
The commands I checked are M119, M122, M500, M502 and M115 (output in the first post) and all worked well.

amad3v, 2022-09-20:
The serial speed is the same on both. I tried 115200 and 250000

radek8, 2022-09-20:
I asked about the response to the M503 order

amad3v, 2022-09-20:
I know you did. I didn't try it and I'm assuming it is working like other commands but I'll give it a go

amad3v, 2022-09-20:
M503 output
```
SENDING:M503
echo:; Linear Units:
echo:  G21 ; (mm)
echo:; Temperature Units:
echo:  M149 C ; Units in Celsius
echo:; Filament settings (Disabled):
echo:  M200 S0 D1.75
echo:; Steps per unit:
echo:  M92 X80.00 Y80.00 Z400.00 E500.00
echo:; Max feedrates (units/s):
echo:  M203 X300.00 Y300.00 Z5.00 E25.00
echo:; Max Acceleration (units/s2):
echo:  M201 X3000.00 Y3000.00 Z100.00 E10000.00
echo:; Acceleration (units/s2) (P<print-accel> R<retract-accel> T<travel-accel>):
echo:  M204 P3000.00 R3000.00 T3000.00
echo:; Advanced (B<min_segment_time_us> S<min_feedrate> T<min_travel_feedrate> J<junc_dev>):
echo:  M205 B20000.00 S0.00 T0.00 J0.01
echo:; Home offset:
echo:  M206 X0.00 Y0.00 Z0.00
echo:; Material heatup parameters:
echo:  M145 S0 H180.00 F0
echo:  M145 S1 H240.00 F0
echo:; Hotend PID:
echo:  M301 P22.20 I1.08 D114.00
echo:; Stepper driver current:
echo:  M906 X800 Y800 Z800
echo:  M906 I1 Z800
echo:  M906 T0 E800
echo:; StallGuard threshold:
echo:  M914 X125 Y125
echo:; Driver stepping mode:
echo:  M569 S1 X Y Z
echo:  M569 S1 I1 Z
echo:  M569 S1 T0 E
```

radek8, 2022-09-20:
So that's okay too.
Can you try the display with a different printer?

amad3v, 2022-09-20:
Sorry, that's the only one I have

radek8, 2022-09-20:
Did you check the cable to see if any wire is broken?

The wiring appears to be connected correctly.
You say that the serial port speed is set to the same on both devices.
M503 is enabled.
You have set serial support 1 on Merlin (for SKR3)

So the display should connect...
Looks like a faulty serial port pin. If it was a bad pin, it could be remapped to another pin, but FW modification will be required.
Do you want to try it?

amad3v, 2022-09-20:
As I told you earlier the wires connectivity is fine, no broken wires.

I'm willing to try it but how does that play with the warranty? I just got the boards, so if the board is faulty, it is better to replace it

amad3v, 2022-09-20:
Are you with BTT?

radek8, 2022-09-20:
I'm not from BTT.
This is not a HW intervention in the board, so it will not affect the warranty.
Could be a faulty pin on the SKR or TFT.
If you're going to make a claim, you should know what's wrong.

radek8, 2022-09-20:
If you want an answer from BTT, go to their website.
BTT support is not responding on github.

radek8, 2022-09-20:
Both SKR and TFT have several serial ports.
If it happens that a pin of this port is defective (this has already been addressed here several times), another pair of pins can be used.
In case of a faulty pin on the SKR, the relevant port must be enabled in Marlin.
In case of a faulty pin on the TFT, the pin must be remapped in the FW of the display.
Then just adjust the cable and connect the appropriate pins.

read here
https://github.com/bigtreetech/BIGTREETECH-TFT35-V3.0/issues/120

amad3v, 2022-09-21:
Good to know. Thanks!

Should I follow your comment [here](https://github.com/bigtreetech/BIGTREETECH-TFT35-V3.0/issues/120#issuecomment-791666846)?

radek8, 2022-09-21:
Yes
I'm at work now.
In the evening I will write you the procedure for remapping the pins to SKR or TFT.
Compiling your own Marlin?

amad3v, 2022-09-21:
Much appreciated
Yes I'm compiling marlin

radek8, 2022-09-21:
Do everything I tell you.

Compile Marlin with a serial communication speed of 115200
and enable serial port 3 for wifi
The configuration.h file in Marlin
#define BAUDRATE 115200
#define SERIAL_PORT_3 3
Upload Marlin to SKR.

First, try to remap the serial ports to TFT:
Upload the attached FW for TFT + ini file to the TFT.
If successful, the files will be renamed to .CUR
From the serial connector on the TFT side, disconnect the 2 wires leading to RX and TX
connect these wires to the RX3 and TX3 pins in the UART3 connector

Turn on the printer.

adjusted FW for TFT:

[BIGTREE_GD_TFT35_V3.0_E3.27.x.zip](https://github.com/bigtreetech/SKR-3/files/9618967/BIGTREE_GD_TFT35_V3.0_E3.27.x.zip)

[image]

If the pins on the TFT were faulty the Printer should communicate with the TFT

amad3v, 2022-09-21:
Thank you, will try it.
Did you redefine USART2_TX_PIN and USART2_RX_PIN as PB10 and PB11 in the display's firmware?

radek8, 2022-09-25:
Is there anything new with your problem?

amad3v, 2022-09-28:
Hi radek8,

I've been away for work, didn't get the chance to try it.
Will let you know as soon as I try it.

amad3v, 2022-10-01:
I managed to get it working.
Thanks for your help @radek8

radek8, 2022-10-02:
So what was the problem?

GreaterJoe, 2022-12-29:
> I managed to get it working. Thanks for your help @radek8

@amad3v How?

EricH-DW, 2024-07-08:
Just got done with this issue but on a Manta E3EZ. I solved it by enabling all 3 serial ports. I searched online for hours but could not find the answer so to help anyone else out who is having this issue here's how I got the TFT35 v3.0.1 working on my Manta E3EZ:

Note this build used Marlin-2.1.2.4. Also note that there may be a better way to do this in Marlin but "better" is the enemy of "good enough"

The hint on how to get the serial working came from the file pins_BTT_MANTA_E3_EZ_V1_0.h. The comment on line 250 is the one.

Ports PD8 and PD9 are driven by USART #3 (MCU Datasheet) so that port is the one you need. I tried setting port 1 to 3 but that didn't work. I went back to all three and it worked again. Here is the relevant section of Configuration.h (also attached both configs in the zip):
/**
 * Select the serial port on the board to use for communication with the host.
 * This allows the connection of wireless adapters (for instance) to non-default port pins.
 * Serial port -1 is the USB emulated serial port, if available.
 * Note: The first serial port (-1 or 0) will always be used by the Arduino bootloader.
 *
 * :[-1, 0, 1, 2, 3, 4, 5, 6, 7]
 */
#define SERIAL_PORT -1
#define PIN_SERIAL_RX           PD9
#define PIN_SERIAL_TX           PD8
/**
 * Serial Port Baud Rate
 * This is the default communication speed for all serial ports.
 * Set the baud rate defaults for additional serial ports below.
 *
 * 250000 works in most cases, but you might try a lower speed if
 * you commonly experience drop-outs during host printing.
 * You may try up to 1000000 to speed up SD file transfer.
 *
 * :[2400, 9600, 19200, 38400, 57600, 115200, 250000, 500000, 1000000]
 */
#define BAUDRATE 250000

//#define BAUD_RATE_GCODE     // Enable G-code M575 to set the baud rate

/**
 * Select a secondary serial port on the board to use for communication with the host.
 * Currently Ethernet (-2) is only supported on Teensy 4.1 boards.
 * :[-2, -1, 0, 1, 2, 3, 4, 5, 6, 7]
 */
#define SERIAL_PORT_1 1
#define BAUDRATE_1 115200   // :[2400, 9600, 19200, 38400, 57600, 115200, 250000, 500000, 1000000] Enable to override BAUDRATE

/**
 * Select a secondary serial port on the board to use for communication with the host.
 * Currently Ethernet (-2) is only supported on Teensy 4.1 boards.
 * :[-2, -1, 0, 1, 2, 3, 4, 5, 6, 7]
 */
#define SERIAL_PORT_2 2
#define BAUDRATE_2 115200   // :[2400, 9600, 19200, 38400, 57600, 115200, 250000, 500000, 1000000] Enable to override BAUDRATE

/**
 * Select a third serial port on the board to use for communication with the host.
 * Currently only supported for AVR, DUE, LPC1768/9 and STM32/STM32F1
 * :[-1, 0, 1, 2, 3, 4, 5, 6, 7]
 */
#define SERIAL_PORT_3 3
#define BAUDRATE_3 115200   // :[2400, 9600, 19200, 38400, 57600, 115200, 250000, 500000, 1000000] Enable to override BAUDRATE

[Configuration.zip](https://github.com/user-attachments/files/16128555/Configuration.zip)

Hopefully this helps. Cheers, Eric