bigtreetech/SKR-3 #25: SKR 3 ez BLTouch fails to power on or funtion

RandieBarsteward, 2022-09-22:
I have just upgraded to a SKR 3 ez on my Ender 5 Plus and am having issues with my BLTouch v3.1 which does not power on nor respond to any commands. The BLTouch v3.1 in question works on another printer without issue and I have also tested a second BLTouch which also does not function.

The console throws an error which reads: "BLTouch failed to verify sensor state"

My Klipper config can be found [[HERE](https://github.com/RandieBarsteward/Ender-5-Plus-Klipper/blob/main/printer.cfg%20SKR%203%20EZ) (I have updated this with recent changes)]
[image]

[image]

My BLTouch wiring can be seen above and appears to match the wiring diagram in the manual.

I am starting to suspect a defective board, can anyone sanity check this issue for me?

RandieBarsteward, 2022-09-25:
Still don't know what the issue was but I reflashed the firmware, removed all wires and started again... And it's working.

closed 2022-09-25 by RandieBarsteward

pixel-shock, 2022-10-05:
Hi, I have the same issue. My BLTouch worked before and I did not change any wirings ... switched from the 1.4Turbo to the EZ and upgraded the FW with the pinout given in the schema and in the FW examples ...

pixel-shock, 2022-10-05:
OK, my fault, I set the PIN to the wrong one ...  sorry for that :-/