bigtreetech/SKR-3 #26: SKR 3 no communication with TMC2209

Drzet, 2022-09-23:
Upgrading from SKR 1.4. TMC2209 on all axes. Installed all jumpers according to manual.

Sensorless homing not working.

Send: M122
Recv: 		X	Y	Z	E	E1
Recv: Address		0	0	0	0	0
Recv: Enabled		false	false	false	false	false
Recv: Set current	800	800	800	800	800
Recv: RMS current	1436	1436	1436	1436	1436
Recv: MAX current	2025	2025	2025	2025	2025
Recv: Run current	25/31	25/31	25/31	25/31	25/31
Recv: Hold current	12/31	12/31	12/31	12/31	12/31
Recv: CS actual	0/31	0/31	0/31	0/31	0/31
Recv: PWM scale
Recv: vsense		0=.325	0=.325	0=.325	0=.325	0=.325
Recv: stealthChop	false	false	false	false	false
Recv: msteps		256	256	256	256	256
Recv: interp		false	false	false	false	false
Recv: tstep		0	0	0	0	0
Recv: PWM thresh.	1756	1756	10541	3077	3077
Recv: [mm/s]		90	90	3	30	30
Recv: OT prewarn	false	false	false	false	false
Recv: triggered
Recv:  OTP		false	false	false	false	false
Recv: pwm scale sum	0	0	0	0	0
Recv: pwm scale auto	0	0	0	0	0
Recv: pwm offset auto	0	0	0	0	0
Recv: pwm grad auto	0	0	0	0	0
Recv: off time	0	0	0	0	0
Recv: blank time	16	16	16	16	16
Recv: hysteresis
Recv:  -end		-3	-3	-3	-3	-3
Recv:  -start		1	1	1	1	1
Recv: Stallguard thrs	80	70	0	0	0
Recv: uStep count	0	0	0	0	0
Recv: DRVSTATUS	X	Y	Z	E	E1
Recv: sg_result	0	0	0	0	0
Recv: stst		*	*	*	*	*
Recv: olb
Recv: ola
Recv: s2gb
Recv: s2ga
Recv: otpw
Recv: ot
Recv: 157C
Recv: 150C
Recv: 143C
Recv: 120C
Recv: s2vsa
Recv: s2vsb
Recv: Driver registers:
Recv: 		X	0x00:00:00:00	 Bad response!
Recv: 		Y	0x00:00:00:00	 Bad response!
Recv: 		Z	0x00:00:00:00	 Bad response!
Recv: 		E	0x00:00:00:00	 Bad response!
Recv: 		E1	0x00:00:00:00	 Bad response!
Recv: 
Recv: 
Recv: Testing X connection... Error: All LOW
Recv: Testing Y connection... Error: All LOW
Recv: Testing Z connection... Error: All LOW
Recv: Testing E connection... Error: All LOW
Recv: Testing E1 connection... Error: All LOW

JoaquinBerrios, 2022-09-24:
What sources are you building from, and have you seen these issues?
https://github.com/bigtreetech/SKR-3/issues/13
https://github.com/MarlinFirmware/Marlin/issues/24703

Should be fixed if you build using the latest Marlin 2.1 with patches, or just edit your PlatformIO config files as noted in the issues linked above.

Drzet, 2022-09-24:
First of all, thank! 
**platform_packages = framework-arduinoststm32@4.20200.220530**

I was using the latest 2.1.1 and was not expecting to deal with outdated packages.

closed 2022-09-24 by Drzet