bigtreetech/SKR-3 #29: skr 3 with ez2209 endstop issues

dannikka, 2022-10-04:
I have a skr 3 with tmc2209 driver in uart mode and  sensorless  homing and  all works fine, so i decided to upgrade my drivers to ez2209 with ez driver connectors for better heat dissipation and after installing the new drivers my printers was homing in the wrong direction.
what i have worked out is the end stops are showing triggered with the ez2209 but open with the tmc2209 ?????

bamhm182, 2022-10-30:
I am trying to figure this out as well. It looks like the pinout between a TMC2209 and the EZ-Driver Controller is different. Somehow it must be possible to say "The diag pin is actually over here" but I don't know how. I will post back here if I figure something out.

markbastiaans, 2023-04-23:
The EZ-Driver Connector board does not have the DIAG pin at the expected place. I don't even think it will fit as the driver slot overlaps the place where the header will go. Hopefully BTT will fix this with a new Connector board release.

I have solved this by soldering jumper cables to the DIAG pin and directly routing them to the X and Y endstop min pins.

[image]

vovodroid, 2024-03-30:
I've purchased SKR3+EZ2209 in November 2023, still doesn't working.

@markbastiaans , besides soldering, no other changes from standard settings were needed?

vovodroid, 2024-03-30:
@dannikka , did you succeed getting it working?

vovodroid, 2024-03-31:
@markbastiaans, I followed your solution - it works, thanks!