bigtreetech/SKR-3 #30: pt100

NastyPastie, 2022-10-12:
Hi I'm looking into using the PT100 with the SKR3. I cannot find anywhere for this compatability between the skr3 and pt100's. I know that the PT1000 can be plugged directly into this board and in the firmware you can set this but is it possible to plug the pt100 into this port and change any values to allow this to work correctly.

Thanks

Abe8816, 2022-10-29:
Yes same here I thought I would need a pt100 amplifier.

Voxel3DPrinting, 2023-06-23:
Did you find this out?

dbuezas, 2023-10-21:
I'll attempt to plug 2 pt100 to an skr 3 board.
I know it's a different processor, but this guide for the 1.4 seems like a reasonable starting point: https://github.com/GadgetAngel/MAX31865-SKR-V1-4-TURBO-GUIDE