bigtreetech/SKR-3 #31: Error SKR 3 EZ TFT50 Marlin 2.1.1

Sergeykurm, 2022-11-01:
Hello! Tell me Why X writes ALL LOW. Rearranged the driver in places, but writes an error on X?

Haven't been able to connect to RS232 TFT50 yet. No connection.
I had to connect to UART3. Works but does not print from SD card SKR 3 EZ. He sees the card, writes that the printing has begun, but there is no heating.
From Pronterface everything prints well and sees the map. But it gives an error M122 X ALL LOW.

  Prints well from the SD TFT screen.
When all wires are connected for Marlin mode, the sound from the screen disappears.

Please help set up!

PS Foto https://www.dropbox.com/scl/fo/vnacc8recw12cl2iijdbn/h?dl=0&rlkey=3gpb1whn6d95tbdiy4e67ixs9

Sergeykurm, 2022-11-01:
ALL LOW solved with this post https://github.com/MarlinFirmware/Marlin/issues/24703

Sergeykurm, 2022-11-01:
The second issue has been resolved. The heaters didn't turn off. On the screen in the settings I found: cancel gcode - on.
Or in config.ini cancel_gcode_enabled:1

radek8, 2022-11-01:
If you want to use UART3 to connect to SKR, you have to modify and compile your own FW for TFT

Sergeykurm, 2022-11-02:
> If you want to use UART3 to connect to SKR, you have to modify and compile your own FW for TFT
> 
> [image]
 
Yes, i modity FW as in the example.

radek8, 2022-11-02:
This is not an example, this is a specific change you have to make before compiling the FW for the TFT display

Sergeykurm, 2022-11-03:
> This is not an example, this is a specific change you have to make before compiling the FW for the TFT display

This is clear. Question Why is RS232 not working?

radek8, 2022-11-03:
It looks like you have a faulty pin on the RS232 connector on the processor. Fortunately, the communication can be remapped to another serial input.
You are not the first with this problem.

closed 2022-11-03 by Sergeykurm