bigtreetech/SKR-3 #33: Bottom silk screen and schematic for motor driver disagree.

BrashEndeavours, 2022-11-03:
Ordering of connection on schematics () of the  E1DM 2x8 pin connector does not match the pin ordering on the bottom silkscreen. 

The bottom silk screen shows:
EN M0 M1 M2 RST SLP STP DIR 
The schematic shows:
EN M1 M2 M3 RST SLO STP DIR

additionally, the other 8 pin connector bottom silk screen shows:
VM GND B1 B2 A2 A1 VDD GND
Whereas the schematic shows the connector is:
VT G 2B 2A 1A 1B V G

Which is correct? Please and thank you.

HTMLdu, 2023-06-22:
I believe that the schematic is correct for the second one but plug it in and if there is an error or the motors don't move properly try the silkscreen

unstable-deadlock, 2024-04-06:
Okay so basically what my issue raised.
Thank you for your input.