bigtreetech/SKR-3 #36: Doesnt build: variant_MARLIN_STM32H743VX.h: No such file or directory

KittyDotNet, 2022-12-05:
I have a fresh install of vscode/platformio, and a fresh download of this repo, no changes. i hit build, and it says `fatal error: variant_MARLIN_STM32H743VX.h: No such file or directory`

i cant find any reference to this file at all

GD485, 2022-12-05:
You need the bug fix version of marlin.  It’s in  the .zip file on the SKR-3 repository. The most recent version of Marlin do not support your processor.

KittyDotNet, 2022-12-05:
Its bugfix 2.0.9.3, from this

growwheels, 2022-12-09:
I'm having the same issue..... using bugfix 2.0.9.3 on a windows pc

aleixq, 2023-01-03:
Look at https://github.com/bigtreetech/SKR-3/issues/13#issuecomment-1213147119 , I have renamed the file to variant_MARLIN_STM32H743VX.h:

`mv ./buildroot/share/PlatformIO/variants/MARLIN_H743Vx/variant_MARLIN_STM32H743Vx.h ./buildroot/share/PlatformIO/variants/MARLIN_H743Vx/variant_MARLIN_STM32H743VX.h`

And can compile