bigtreetech/SKR-3 #39: Unable to get USB connection on SKR 3 (Marlin)

Yinameah, 2023-01-26:
Hello,

I just received my board. Successfully compile the Marlin firmware.

I copy `firmware.bin` on the SD card, and power up the board. The power led turn on, and the status led flashes.

However, I don't get any USB device reported on my computer. I checked the CAN/USB switch, it's in correct position. On the SD card, the `FIRMWARE.CUR` file appear correctly.

I have no idea what to try next. Any idea ?

Yinameah, 2023-01-26:
By the way, if I hold the BOOT button, and then press the RESET button, I get this with dmesg
```
[ 3797.405206] usb 1-2: new full-speed USB device number 6 using xhci_hcd
[ 3797.545424] usb 1-2: not running at top speed; connect to a high speed hub
[ 3797.547305] usb 1-2: New USB device found, idVendor=0483, idProduct=df11, bcdDevice= 2.00
[ 3797.547319] usb 1-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 3797.547325] usb 1-2: Product: DFU in FS Mode
[ 3797.547330] usb 1-2: Manufacturer: STMicroelectronics
[ 3797.547334] usb 1-2: SerialNumber: 306535463231
```
So clearly, the USB is working, just when the firmware is loaded, the Serial via USB interface is not configured properly

Yinameah, 2023-01-26:
Oh, just found out this : 

If I set the device in DFU mode (as shown in the last comment), and then try to upload the firmware directly via platformio, it starts well : 

```
Opening DFU capable USB device...
Device ID 0483:df11
Device DFU version 011a
Claiming USB DFU Interface...
Setting Alternate Interface #0 ...
Determining device status...
DFU state(2) = dfuIDLE, status(0) = No error condition is present
DFU mode device DFU version 011a
Device returned transfer size 1024
DfuSe interface name: "Internal Flash   "
Downloading element to address = 0x08020000, size = 216392

Erase           [                         ]   0%            0 bytes
Erase           [                         ]   0%            0 bytes
Erase           [=                        ]   4%         9216 bytes
.....
```
and seems to work fine until if crashes on this : 
```
...
Download        [======================== ]  96%       207872 bytes
Download        [======================== ]  97%       210944 bytes
Download        [=========================] 100%       216392 bytes
Download done.
File downloaded successfully
Submitting leave request...
dfu-util: Error during download get_status
*** [upload] Error 74
```

Yinameah, 2023-01-28:
Ok, never mind.

I spent waay to much time on this, before realizing that it's written STM32H723 on the mcu present on my board. Turns out that it's supposed to be the STM32H743, and that the UBS peripheral is one of the differences between the two mcu. 

So it's not possible to have the serial over USB working.

I'm a bit surprised they managed to assemble a board with the wrong mcu... Would not recommend to buy.

Closing the issue since it's not a software issue.

closed 2023-01-28 by Yinameah

infowolfe, 2023-03-18:
That's fucking hilarious. Mine's also the H723 instead of the H743.

Yinameah, 2023-03-18:
I reach out to the seller, that said it was a normal iteration with no impact. Which is... well, debatable.

However, I did managed to burn klipper firmware at the end, using a st-link.

(did not managed to make Marlin work)

Turned out I find klipper a super cool project, and I'm now happy the way it went.

If you want to go down this way and experience issues, let me know, I might give you a few tips.

infowolfe, 2023-03-18:
@Yinameah I've attempted flashing klipper on both dfu and on sdcard and nothing doing. I have no USB port unless I'm in dfu mode. And yeah. if even their own firmware-USB.bin doesn't work... wellllll that "no impact" is a bold-faced lie.

Yinameah, 2023-03-18:
Yeah, I know, same here. that's why I ended up using a st-link v2. One of these chip ones you find everywhere

infowolfe, 2023-03-18:
Could you document how you did it and any settings you used? I'd prefer to
not have to return this board since it's installed.

HTMLdu, 2023-03-26:
@Yinameah Did you run into any issues getting marlin to work other than what is posted?

Yinameah, 2023-03-26:
@HTMLdu Not sure I get it.

I did NOT manage not have Marlin working. Not even using the st-link v2. I belive Marlin support the H745 and not the H723.

HTMLdu, 2023-03-26:
@Yinameah Ahh I see now. Also I have an explanation for why Klipper works. Klipper does not use serial over usb like Marin does. To use serial in marlin, the pins that the usb port is connected to on the mcu must support serial which on the H723 sadly does not. I am pretty sure that if you could configure software serial it would work though.

SomeUnknownUser, 2023-04-02:
> @Yinameah To use serial in marlin, the pins that the usb port is connected to on the mcu must support serial which on the H723 sadly does not. 

Wait… that means it’s not possible to get any USB COM interface with a H723 because of a hardware limitation?
How to get pronterface etc running?

Yinameah, 2023-04-02:
Well. Technically it is more of a software issue, because the USB to serial link works in the klipper firmware.

Again, I was personally not to able to have it working with Marlin, and don't plan to look further. Might be that some software serial configuration works, might be that a new motherboard should be defined. It's definitely possible to do in theory. 

But yes, as far as I can tell, the H723 version is NOT compatible with Marlin at the current state of things.

SomeUnknownUser, 2023-04-02:
Okay thank you!
That’s horrible! Mostly because BTT support do not answer any support mails. Not even if you directly buy hardware on the website.

That’s my very last Bigtreetech Hardware.

Yinameah, 2023-04-02:
Yes, it's sad.

You might open a ticket in Marlin repo. Someone who knows both stm32 chip involved might be able to provide a fix.

But if you have a raspberry lying around, and a st-link programmer, I would recommand give klipper a try. (follow instruction in the thread referenced above) 
I was not convinced by the kipper approach, but turns out I find it a great project, with amazing documentation.

SomeUnknownUser, 2023-04-02:
Thanks a lot!
Unfortunately I don’t have a pi lying around - and a new one is more expensive then a other mainboard.
But even a other mainboard is to expensive to me right now.

So let’s hope that someone is coding a solution for that somewhen.

HTMLdu, 2023-04-02:
The new bugfix branch fixed this issue compile with the H723 environment instead of H743

SomeUnknownUser, 2023-04-02:
I did 2 days before and the board itself is running. Everything works (including TFT serial) etc. but the USB COM does not work. Absolutely nothing happens when connecting the usb with a PC

HTMLdu, 2023-04-02:
Hmm it worked for me after flash the new firmware. Are you 100 sure that the usb/can button is in the up position? Also, have you set serial port to -1?

SomeUnknownUser, 2023-04-02:
Hmm interesting…

yes button is in up position. Also pushed the button several times do double and tripple check.

It looks like I defined -1:

`#define SERIAL_PORT_2 `
`#define BAUDRATE_2 115200   // :[2400, 9600, 19200, 38400, 57600, 115200, 250000, 500000, 1000000] Enable to override BAUDRATE
`

SERIAL 1 is used for the TFT

anything I forgot?

HTMLdu, 2023-04-03:
I don't have a tft but could you try swapping the serial ports even if the tft does not work it is still good for troubleshooting. Also make sure you are using a good usb cable and baud rate on your host is set correctly

SomeUnknownUser, 2023-04-03:
Thank you! I don’t know what happened but it works now. I swapped SERIAL and SERIAL_PORT_2. It was working fine. After that I changed back and it’s also working. Typo? Who knows.

rodrigo2019, 2023-11-11:
Same issue for me, my board came with 723 instead 743, luckly we have guys like @Yinameah, and also for my luck i'm using klipper instead marlin. So recompiling the firmware changing it to the stm final 723 solved my problem