bigtreetech/SKR-3 #40: Display doesn't work (DZ2004A)

Monokel777, 2023-01-29:
My Display (Type: DZ2004A) form the Geeetech Prusa I3 Pro X doesn´t start even the Backlight with Marlin 2.1.2.

I Installed the SKR 3.0 (no EZ) on my Geeetech and the Display wouldn`t show something.

1. I compliled marlin several times with different Display-types (REPRAP_DISCOUNT_SMART_CONTROLLER, REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER),
2. switched the cables
3. tried to take my old GT2560, plugged the LCD-Wire in. (worked, so the Display seems to be fine)

Now I think the Pinout May be faulty. Do you have an advice?

You will find my Confiuration.h in the attatchments

Monokel777, 2023-01-29:
[Configuration.txt](https://github.com/bigtreetech/SKR-3/files/10530692/Configuration.txt)

Monokel777, 2023-01-30:
```
              ------                                   ------
 (BEEPER) PC5  | 1  2 | PB0  (BTN_ENC)  (MISO)      PA6 | 1  2 | PA5 (SCK)
 (LCD_EN) PB1  | 3  4 | PE8  (LCD_RS)   (BTN_EN1)   PE7 | 3  4 | PA4 (SD_SS)
 (LCD_D4) PE9  | 5  6   PE10 (LCD_D5)   (BTN_EN2)   PB2 | 5  6   PA7 (MOSI)
 (LCD_D6) PE11 | 7  8 | PE12 (LCD_D7)   (SD_DETECT) PC4 | 7  8 | RESET
            GND | 9 10 | 5V                          GND | 9 10 | --
                ------                                   ------
                 EXP1                                     EXP2
 
#define EXP1_01_PIN                         PC5
#define EXP1_02_PIN                         PB0
#define EXP1_03_PIN                         PB1
#define EXP1_04_PIN                         PE8
#define EXP1_05_PIN                         PE9
#define EXP1_06_PIN                         PE10
#define EXP1_07_PIN                         PE11
#define EXP1_08_PIN                         PE12

#define EXP2_01_PIN                         PA6
#define EXP2_02_PIN                         PA5
#define EXP2_03_PIN                         PE7
#define EXP2_04_PIN                         PA4
#define EXP2_05_PIN                         PB2
#define EXP2_06_PIN                         PA7
#define EXP2_07_PIN                         PC4
#define EXP2_08_PIN                         -1

```

DenDeg, 2023-01-31:
Maybe it's because this old display use 5V TTL levels. Modern MCU use 3.3V TTL levels.

Monokel777, 2023-01-31:
@DenDeg could I change the level in Marlin?

Thank you for your response!

HTMLdu, 2023-03-27:
@Monokel777 No you cannot. The MCU physicaly operates on 3.3 volts so you cannot change it.

HTMLdu, 2023-03-27:
The Reprapdisount display and ones that use the same pinout need to have the VCC and GND pins swapped