bigtreetech/SKR-3 #44: SKR3 (MCU723) can flash fw. but USB no communication

sacy80, 2023-02-17:
i got a SKR3 with a diff MCU (STM32H723VGT6). look like Marlin Firmware can load, as firmware.bin has change to firmware.CUR. But USB communication with PC are not alive, no COM Port detected. Any one has this issue.

Platform.ini-
platformio]
src_dir      = Marlin
boards_dir   = buildroot/share/PlatformIO/boards
default_envs = STM32H743Vx_btt
include_dir  = Marlin
extra_configs =
    Marlin/config.ini
    ini/avr.ini
    ini/due.ini
    ini/esp32.ini
    ini/features.ini
    ini/lpc176x.ini
    ini/native.ini
    ini/samd21.ini
    ini/samd51.ini
    ini/stm32-common.ini
    ini/stm32f0.ini
    ini/stm32f1-maple.ini
    ini/stm32f1.ini
    ini/stm32f4.ini
    ini/stm32f7.ini
    ini/stm32h7.ini
    ini/stm32g0.ini
    ini/teensy.ini
    ini/renamed.ini

Config.h-
// Choose the name from boards.h that matches your setup
#ifndef MOTHERBOARD
  #define MOTHERBOARD BOARD_BTT_SKR_V3_0
#endif

#define SERIAL_PORT -1

/**
 * Serial Port Baud Rate
 * This is the default communication speed for all serial ports.
 * Set the baud rate defaults for additional serial ports below.
 *
 * 250000 works in most cases, but you might try a lower speed if
 * you commonly experience drop-outs during host printing.
 * You may try up to 1000000 to speed up SD file transfer.
 *
 * :[2400, 9600, 19200, 38400, 57600, 115200, 250000, 500000, 1000000]
 */
#define BAUDRATE 250000

//#define BAUD_RATE_GCODE     // Enable G-code M575 to set the baud rate

/**
 * Select a secondary serial port on the board to use for communication with the host.
 * Currently Ethernet (-2) is only supported on Teensy 4.1 boards.
 * :[-2, -1, 0, 1, 2, 3, 4, 5, 6, 7]
 */
#define SERIAL_PORT_2 1
#define BAUDRATE_2 250000   // :[2400, 9600, 19200, 38400, 57600, 115200, 250000, 500000, 1000000] Enable to override BAUDRATE

ericc0922, 2023-02-23:
I got the same issue. Do you solved?

sacy80, 2023-02-23:
nope, still waiting for the pull request to review and merged to bugfix main.

https://github.com/MarlinFirmware/Marlin/pull/25387

CristianoGorla, 2023-02-23:
> nope, still waiting for the pull request to review and merged to bugfix main.
> 
> [MarlinFirmware/Marlin#25387](https://github.com/MarlinFirmware/Marlin/pull/25387)

You haven't to wait, you can apply the PR and compile it. I did it and it's working. The only problem are the TMC2209, you have to use a specific STM32 version, I modified the stm32h7.ini, env:STM32H723Vx_btt:
platform_packages           = framework-arduinoststm32@>4.20000.210603 ; TO AVOID TMC CONNECTION ERROR
board_build.f_cpu           = 550000000L ; H723 MCU 550Mhz

Processing STM32H723Vx_btt (platform: ststm32@~14.1.0; board: marlin_STM32H723Vx; framework: arduino)
--------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/marlin_STM32H723Vx.html
PLATFORM: ST STM32 (14.1.0) > STM32H723VE (564k RAM. 512k Flash)
HARDWARE: STM32H723VET6 550MHz, 564KB RAM, 512KB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, jlink, stlink)
PACKAGES: 
 - framework-arduinoststm32 @ 4.20200.221104 (2.2.0) 
 - framework-cmsis @ 2.50700.210515 (5.7.0) 
 - toolchain-gccarmnoneeabi @ 1.90201.191206 (9.2.1)

thisiskeithb, 2023-02-24:
> The only problem are the TMC2209, you have to use a specific STM32 version, I modified the stm32h7.ini, env:STM32H723Vx_btt:
platform_packages = framework-arduinoststm32@>4.20000.210603 ; TO AVOID TMC CONNECTION ERROR

I've updated ststm32 platform & framework to match the original SKR 3 environment.

> board_build.f_cpu = 550000000L ; H723 MCU 550Mhz

The cpu speed is already set in [PIO's STM32H723Vx board json file](https://github.com/MarlinFirmware/Marlin/blob/18db956f40c9f61a63b712f5fd9ae75aa248451c/buildroot/share/PlatformIO/boards/marlin_STM32H723Vx.json#L6), so you don't need to add this:

```json
  "build": {
    "core": "stm32",
    "cpu": "cortex-m7",
    "extra_flags": "-DSTM32H7xx -DSTM32H723xx",
    "f_cpu": "550000000L",
    "mcu": "stm32h723vet6",
    "product_line": "STM32H723xx",
    "variant": "MARLIN_H723vx"
  }
```

sacy80, 2023-02-24:
Nice. Will try later. Thanks alot

ericc0922, 2023-02-24:
Follow it but got Error:
Error: Build environment 'STM32H723Vx_btt' is incompatible with BOARD_BTT_SKR_3. Use one of these: STM32H743Vx_btt
Any comments? Thanks alot

thisiskeithb, 2023-02-24:
> Error: Build environment 'STM32H723Vx_btt' is incompatible with BOARD_BTT_SKR_3. Use one of these: STM32H743Vx_btt

You're not using all the updated files from my PR since the `env:STM32H723Vx_btt` was added as a compatible environment to the SKR 3 board/pins.

Ensure you're using https://github.com/thisiskeithb/Marlin/tree/pr/octopus_max_ez ([.ZIP link](https://github.com/thisiskeithb/Marlin/archive/refs/heads/pr/octopus_max_ez.zip)) until it is merged  into `bugfix-2.1.x`.

ericc0922, 2023-02-24:
I tried. The same problem.
'Error: Build environment 'STM32H723Vx_btt' is incompatible with BOARD_RAMPS_14_EFB. Use one of these: mega2560, mega1280'

If I change board name to 'BTT_SKR_V3_0'. It comes:
'error: #error "Unknown MOTHERBOARD value set in Configuration.h."'

thisiskeithb, 2023-02-24:
> I tried. The same problem.
> 'Error: Build environment 'STM32H723Vx_btt' is incompatible with BOARD_RAMPS_14_EFB. Use one of these: mega2560, mega1280'

That is expected because you have the incorrect motherboard defined.

> If I change board name to 'BTT_SKR_V3_0'. It comes:
> 'error: #error "Unknown MOTHERBOARD value set in Configuration.h."'

You should be using `BOARD_BTT_SKR_V3_0`, not `BTT_SKR_V3_0`.

ericc0922, 2023-02-24:
Working now. Thanks bro. You are my angel!

rzboril, 2023-02-24:
And now foe Marlin 2.1.2 ? Pleas

ericc0922, 2023-03-02:
Now shows TMC2209 V1.3 connection error: all low. All X, Y, Z, E works but this error message.
M122 return:

Driver registers:
		X	0x00:00:00:00	 Bad response!
		Y	0x00:00:00:00	 Bad response!
		Z	0x00:00:00:00	 Bad response!
		E	0x00:00:00:00	 Bad response!
Testing X connection... Error: All LOW
Testing Y connection... Error: All LOW
Testing Z connection... Error: All LOW
Testing E connection... Error: All LOW

How to solve it?? >_<

rzboril, 2023-03-02:
Try physicaly repinned ABCD to ACBD on motor. If only stress motor. And 550Mhz board konfig Octopus Max EZ.

ericc0922, 2023-03-02:
> Try physicaly repinned ABCD to ACBD on motor. If only stress motor. And 550Mhz board konfig Octopus Max EZ.

But the motor works fine, just an error message appears. It's wired.

CristianoGorla, 2023-03-02:
> With BTT TMC2209 V1.3 driver, X Y Z E all works fine. But it has 'All LOW' error message.
> 
> M122 return: Driver registers: X 0x00:00:00:00 Bad response! Y 0x00:00:00:00 Bad response! Z 0x00:00:00:00 Bad response! E 0x00:00:00:00 Bad response! Testing X connection... Error: All LOW Testing Y connection... Error: All LOW Testing Z connection... Error: All LOW Testing E connection... Error: All LOW
> 
> How to solve it?? >_<

I modified the stm32h7.ini, env:STM32H723Vx_btt:
platform_packages = framework-arduinoststm32@>4.20000.210603 ; TO AVOID TMC CONNECTION ERROR

rzboril, 2023-03-02:
Bad drivers ? Try any DRV8825 ?

ericc0922, 2023-03-03:
> did you power the board with 12v or 24v. the tmc will not run if you only power the board over 5v usb.

Yes. I power the board 24V and motor are working fine. But still have All LOW error.

ericc0922, 2023-03-03:
> Bad drivers ? Try any DRV8825 ?

Motor working fine. I think drivers are ok.

bz835, 2023-03-03:
Hi there, I am pretty new to the Marlin Firmware theme. I have a STM32H723 SKR3 and have no idea how to put the needed environment to my Marlin 2.1.2. Firmware. Does someone have a good tutorial to do that task? Thanks for the help in advance.

bz835, 2023-03-03:
I am able to compile it to the stm32h743 version, but the Board does not show any reaction when I put the SD card in and press the restart button.

bz835, 2023-03-03:
Hi there, I managed to get the firmware on the board and the LCD is showing the printer. My last problem is the bl touch. I selected the correct pin for the Probe (PC_13) and it still does not work. I changed the config file according to the setup of a bl touch. The cables are connected in the right order. I have no idea what to do next :S Please help.

thisiskeithb, 2023-03-05:
> I modified the stm32h7.ini, env:STM32H723Vx_btt:
> platform_packages = framework-arduinoststm32@>4.20000.210603 ; TO AVOID TMC CONNECTION ERROR

@CristianoGorla: So `framework-arduinoststm32@~4.20200.220530` doesn't work for the new SKR 3 with STM32H723VGT6? I don't have this new SKR 3 variant, so I can't test here.

sacy80, 2023-03-05:
Notice after awhile tmc has error.  Not sure where goes wrong.

thisiskeithb, 2023-03-11:
https://github.com/MarlinFirmware/Marlin/pull/25387 has been merged upstream in the main Marlin repository, so pull down the latest `bugfix-2.1.x` for Octopus Max EZ & SKR V3/EZ support.

`2.1.3` is expected very Soon:tm:, so you can also wait for that if you're shy about using bugfix.

rzboril, 2023-03-14:
Thank you girls.
I now tested.

rzboril, 2023-03-14:
> > Bad drivers ? Try any DRV8825 ?
> 
> Motor working fine. I think drivers are ok.

And:   /**
   * Enable M122 debugging command for TMC stepper drivers.
   * M122 S0/1 will enable continuous reporting.
   */
  #define TMC_DEBUG

Enabled ?

dutchronnie, 2023-04-09:
Can somebody write down exactly what i have to do to get this board working?
I don't understand all this comments.

I do have an SKR 3 EZ, firmware flashing is ok, but no connection to printer, Marlin, klipper and TFT screen al the same problem.
On the board there is an STM32H723Vx  chip, instead of an STM32H743Vx

Help i have been trying for a week now, and nothing works.

sacy80, 2023-04-09:
You may want to try marlin bigfix 2.1x

dutchronnie, 2023-04-09:
I have changed the env to the correct mcu, that works a little bit better.

The serial connection is now flashing on, off all the time.

My settings in configuration.h are:

#define MOTHERBOARD BOARD_BTT_SKR_V3_0_EZ
#define SERIAL_PORT -1
#define BAUDRATE 250000
#define SERIAL_PORT_2 1
#define SERIAL_PORT_3 3

dutchronnie, 2023-04-10:
The problem is solved, it turns out that the screen was nor correct connected to the motherboard what causes the on, off flickering of the serial port.

Everything works fine now for marlin.