bigtreetech/SKR-3 #48: Provide support/documentation for STM32H723 variant

infowolfe, 2023-03-18:
If you're going to be slimy and silently change the variant of the MCU being manufactured on your boards, at least document the change with an incremented version number and proper support.

Thanks!

Yinameah, 2023-03-19:
As written in #39, I did managed to use the H723 board version with klipper firmware. Here what I did : 

-  Compile klipper firmware with following setup ( `make menuconfig`):
    - [x] Enable extra low-level configuration options
    - Micro-controller Architecture : STMicroelectronics STM32
    - Processor model : STM32H723
    - Bootloader offset : 128KiB bootloader (SKR SE BX v2.0)
    - Clock Reference : 25 MHz crystal
    - Communication interface : USB (on PA11/PA12)
  
- Get [STM32CubeProgrammer](https://www.st.com/en/development-tools/stm32cubeprog.html)
- Get a compatible SWD programmer. I personally [had one of these around](https://www.aliexpress.com/item/1005003575620794.html?spm=a2g0o.store_pc_allProduct.8148356.3.2b5c7143MdqQVd&pdp_npi=2%40dis%21CHF%21CHF%201.76%21CHF%201.55%21CHF%201.55%21%21%21%21%40211b88f116792176948983072edc12%2112000026345111930%21sh)
 - (Note that my st-link v2 device was not working out-of-the-box, but started to after I hit "firmware upgrade" in the "ST-Link configuration" pannel of STM32CubeProgrammer)
 - Hook the cables to the available onboard SWD interface. (it's next to X-STOP plug). Not sure anymore how I managed to find out the order of the cable, probably by looking at the schematics and assuming that the order of the pin is the same. (CubeProgrammer will get some information about the MCU when correctly connected)
 - Select the `klipper.elf` produced by the klipper build in STM32CubeProgrammer, and hit "Start Programming"

After that, I was able to to connect to the board via USB as it should. Never got to make the "program from SD card" working, but since with klipper you almost never need to re-flash the firmware, it's ok.

I am still not sure why this worked, and not the "program from SD card". Playing with STMCubeMX, you can see that USB peripheral available configuration differ in subtle manner between H723 and H743. I guess you would need to have a slightly modified bootloader, but I won't spend any more time on this, so I'll never know.

infowolfe, 2023-03-19:
@Yinameah thank you for the excellent write-up. It's a shame that BTT/Biqu themselves haven't provided this kind of documentation or incremented the board version number with this silent change.

infowolfe, 2023-03-20:
@Yinameah Thank you... I'll fill in the rest for anybody who comes behind us:

You'll need an ST-Link V2. You'll want to upgrade its firmware. You will need DIO, CLK and GND attached with the board powered from the mains. Use an offset of 0x8020000 as the start of the klipper.elf. When you've done it, run a hard reset and then click "run" in STMCube and if you're watching dmesg on your pi (or usb-attached server) then you'll see this:

```
[2754559.601473] usb 3-4.2: new full-speed USB device number 63 using xhci_hcd
[2754559.793951] usb 3-4.2: New USB device found, idVendor=1d50, idProduct=614e, bcdDevice= 1.00
[2754559.793958] usb 3-4.2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[2754559.793962] usb 3-4.2: Product: stm32h723xx
[2754559.793964] usb 3-4.2: Manufacturer: Klipper
[2754559.793966] usb 3-4.2: SerialNumber: <your-serial-here>
[2754559.806196] cdc_acm 3-4.2:1.0: ttyACM0: USB ACM device
```

HTMLdu, 2023-03-25:
So far has anyone found a way to run Marlin on this board?

infowolfe, 2023-03-25:
@HTMLdu if you're using marlin, I *believe* there's info elsewhere on here on how to flash it with dfu mode after rebuilding. This bug is mostly about Klipper support.

dedekcz, 2023-04-10:
@Yinameah and  @infowolfe Thanks for Your detailed help. When I am following Your way, I can not get any success with Klipper files. But I am able to Flash Marlin 2.1.x with STM32CubeProgrammer. So I suppose that problem might be a klipper.elf file... Would be please possible that You shared Your klipper.elf somewhere?
Thanks a lot.

infowolfe, 2023-04-11:
@dedekcz perfect.

luke6040, 2023-05-04:
@infowolfe Hi, hope you don't mind asking directly, but did you have any issues connecting the st-link? I can't for the life of me get it to connect. I have hooked it up as you described and updated the firmware. 
[image]

infowolfe, 2023-05-04:
@luke6040 You have to go through the st-link firmware update workflow in the blue button before you can actually connect. Also, I'm pretty sure those aren't the connection settings I used, but was a couple months ago.

luke6040, 2023-05-04:
Thanks for the fast reply, I did try the firmware update and seemed to update fine. 
Any idea what settings might be wrong, I have tried aimless changing them with no luck.

infowolfe, 2023-05-04:
@luke6040 sorry man, I'm not sure what else to give you, that was like a month and a half ago that I did mine and I haven't had to do it again since then.

luke6040, 2023-05-04:
No worries, think I will just return it, thanks for the try

infowolfe, 2023-05-04:
@luke6040 one other thing to look into maybe is changing the usb cable you're using. I was able to figure it out, but it's not the most straightforward process.

vfbank, 2024-04-20:
The ST LINK you are using is a Chinese clone. ST is blocking all copied serial numbers as a precaution against clones. If you purchase it as a genuine product, everything will go well.

vfbank, 2024-04-20:
> @infowolfe Hi, hope you don't mind asking directly, but did you have any issues connecting the st-link? I can't for the life of me get it to connect. I have hooked it up as you described and updated the firmware. [image]

The ST LINK you are using is a Chinese clone. ST is blocking all copied serial numbers as a precaution against clones. If you purchase it as a genuine product, everything will go well.

Yinameah, 2024-05-12:
Hello,
I updated klipper today, and I needed to update the firmware as well. Thanks to @vfbank comment, I did not need to plug a ST-LINK and could upgrade via usb. It worked, but for some reason the reset address was not set properly. The `.elf` mentioned 0x0804 while the mcu would start at 0x0800

I changed it here and it worked as expected afterwards
[image]