bigtreetech/SKR-3 #50: Has anyone had issues with a Reprapdiscount full graphic LCD

HTMLdu, 2023-03-27:
Has anyone else had any problems with a Reprapdiscount full graphic LCD not having any display?

Tomzrk, 2023-03-27:
i do , my SKR 3 EZ not communicating with my TFT35 v3

SomeUnknownUser, 2023-03-29:
Same here. But it looks like that nothing seems to work. Even no connection to the board over USB.

so I’m not sure if my board is completely booting or not.

The installation seems to work because I see an FIRMWARE.CUR file

Tomzrk, 2023-03-29:
@SomeUnknownUser Are you using SKR 3 EZ ?

SomeUnknownUser, 2023-03-29:
> @SomeUnknownUser Are you using SKR 3 EZ ?

Yes. Brand new. SKR 3 EZ with TFT35 V3.0.1 and EZ2209 Driver

SomeUnknownUser, 2023-03-29:
> @SomeUnknownUser Can you send me your configuration file

I solved my problem, but I’m not sure if you have the same problem. The result is the same: display does not work.
My board wasn’t booting. You can check that if your PC recognize an COM-port after connecting the usb with the pc. I recommend opening „Device Manager“ on windows and check if something happen if you connect the board. There should came up a new com-port.
I also used pronterface. There should also be a new com-port.
If nothing happens: the board is probably not booting and this results is a black display.

Solution :
New(ish) boards seems to have a new chip and BTT doesn’t seem to have modified the manuals or give somehow and somewhere any informations about that.

The new chip is a STM32H723VGT6.
You need to modify the
platformio.ini file 
and set
default_envs = STM32H723Vx_btt

For that you currently need the bugfix-2.1.x branch or wait for the next marlin 2.1.3 release

it works for me and it just took 5 days and endless hours to get this done. BTT support does not react on support mails :/

Tomzrk, 2023-03-29:
I found it as well but now my problem is one of the Dual Z is working only , trying to see what's wrong with it

HTMLdu, 2023-04-02:
I compiled the new firmware but now everything except the display is working

desmotronic, 2023-04-11:
Same here, running bufix-2.1.x branch.  I just have #define REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER uncommented.  LCD does not power up.  I have tried reversing the cables and also swapping between EXP1/EXP2.  Using 2 ribbon LCD.

Has anyone got their LCD to power up?  If so, what needs to be changed?

desmotronic, 2023-06-11:
Anyone figure this out yet?  I am stumped on how to get this display to work.

desmotronic, 2023-06-21:
Ok following up on this. Turns out the end stop connectors were incorrectly pinned. They were shorting 5V to GND. This caused the entire 5V bus to short as the switches are normally closed. The telltale signs were the LCD was not powering on and neither was the BLTouch. I can confirm LCD works on latest Marlin Bugfix branch without issue.  This ticket should be closed now.

HTMLdu, 2023-06-22:
What I found is that the lcd connectors on either the display or the board are pinned backwards and I needed to cut the notch off and put in the connector backwards