bigtreetech/SKR-3 #52: BTT SKR 3EZ - EZ2209 - ALL HIGH - TMC CONNECTION ERROR

Mydaiel, 2023-03-29:
As soon as I boot the BTT SKR3 EZ board I get a TMC Connection Error reading ALL HIGH on all axis regardless of the number of drivers installed.
I cannot move any of the steppers while this error is present.
I have tried many different Marlin builds and different EZ2209 drivers in different slots.
I have tried all of the solutions in issue #13, #23, #26 threads that bring up the ALL LOW - TMC Connection Issues.
I have tried updating the platform package in the INI file with the below options under the 15.4.1 and 14.1.0:
platform_packages           = framework-arduinoststm32@>4.20000.210603
platform_packages           = framework-arduinoststm32@4.20200.220530
platform_packages           = framework-arduinoststm32@https://github.com/stm32duino/Arduino_Core_STM32/archive/main.zip
I have tried other's firmware files compiled on different computers.
I have been trying new builds with the new nightly marlin updates with no luck.
I have the UART connections pinned, I have tried without those jumpers in with no change.
Short of going and buying another or different board, I am at a loss.
I hate to buy another board if it is a program issue.

Has anyone else struggled with the ALL HIGH error TMC connection issue?

livius2, 2023-04-04:
What do you use on the board? Marlin?

Mydaiel, 2023-04-04:
Hello,

I have been trying the nightly releases of the Marlin 2.1 bug fixes as my
base configuration before updating and compiling in vs-code.

Thank you

livius2, 2023-04-04:
From official repo of marlin?
And did you changed in platform.io default_envs = STM32H743Vx_btt

livius2, 2023-04-04:
try to follow:
https://www.youtube.com/watch?v=r9fsRMXb5X0

And check basic things, like connect power cable to power your motors not only board.
I use it with EZ2209 drivers but with sensorless homing and all works ok.

Mydaiel, 2023-04-04:
I will check out the YouTube video and see if it helps with further troubleshooting. I assume since this is a BTT SKR 3EZ board I am using the EZ2209 drivers since I didn't need an adapter to install them. I am also trying to set up sensor less homing but through UART mode not SPI or anything else since that was what this driver number called for. Maybe I messed up the UART set up part.

livius2, 2023-04-04:
For sensorless homing you must insert small jumper (really small) for every driver.

Mydaiel, 2023-04-08:
On previous builds, I had the small jumpers installed for sensor less homing. The YouTube video you suggested for this build, he did not use sensor less homing so I removed the pins and tried to mimic his configuration just using end stop setup. I still get the ALL HIGH TMC connection error, however, this time I was able to get the steppers to move when using Pronterface software where before when using sensorless homing I couldn't get any response from them at all. Even though they moved, it was not smooth or quiet movement and the Z1 stepper and X stepper started overheating within a minute. I still believe this is some kind of configuration programming error on my part instead of a hardware issue.

Mydaiel, 2023-04-11:
I tried new cables for the steppers and endstops, and new ez2209 drivers. I still get an ALL HIGH Error, The axis stepper motors do move now, but make horrible noise doing so and the endstops do not seem to respond when they reach them. Also the z-axis wants to move up instead of down when homing. 

The steps, acceleration and applied current seem to match the ender 3 pro configuration settings. 

I had to re-wire the stepper motor cables I bought because they were not in the correct configuration. I'm not sure what else is causing all this :/

livius2, 2023-04-11:
Did you power the board and the stepper motors (2 separate powers connetors for the board and for the motors)
Did you removed jumper for power from usb cable?

I suppose this two links should clear what you should setup for sensorless homing
And at all for setup marlin with it.
https://www.youtube.com/watch?v=r9fsRMXb5X0
https://www.youtube.com/watch?v=iwMPuV7TexU

form recent marlin, names of config flags was changed in it, so below newes one.
1. Check how do you setup steeps 
2. You have `INVERT_X_DIR` and same for Y and Z if your steepper go in different direction
3.You can home to min or max by `Y_HOME_DIR`
4. check X_MIN_POS, X_BED_SIZE, ....
5. Check X_MICROSTEPS i use 64 look at X_CURRENT for you motors...
6. look at   #define CHOPPER_TIMING CHOPPER_DEFAULT_24V i have 24v power