bigtreetech/SKR-3 #53: SKR 3 ez no serial connection to computer and to tft screen

dutchronnie, 2023-04-08:
I have been trying for several days now to attach a new SKR 3 ez motherboard to mine CR-10 V2 printer.
Without any success.

Flashing the firmware is ok, but after that i don't get a connection to the computer.
I tried the Marlin Firmware deliverd by Bigtreetech 2.0.9.3.x, not any connection
I tried the original Marlin Firmware 2.1.x, not any connection
Tried to install klipper, also not any connection.

Then i installed marlin 2.1.x again, and now trying to connect with a TFT 28 v3.0, also no connection.

It looks like there is not a single serial port working.

I see many peoply who have connection problems with this board, but see no any solution.
I hope somebody can tell me what to do?

When there is no solution, BigTreeTech is delivering a very bad expensive board.

dutchronnie, 2023-04-10:
Solved:

When installing Marlin, edit the platformio.ini file and change:
default_envs = STM32H743Vx_btt to default_envs = STM32H723Vx_btt
Don't follow the instructions on youtube, bigtreeTech site, that doesn't work for a SKR 3 ez with a STM32H723VGT6 chip

ayankemal, 2023-05-12:
Hi @dutchronnie , I solve this problem today. I'm trying to solve this several days. Please try this , Go to ini folder in Marlin and follow the path. ' ini>stm32h7.ini' . Open this file and change ''DPIN_SERIAL1_RX=PA_9 -DPIN_SERIAL1_TX=PA_10'' to ''DPIN_SERIAL2_RX=PA_9 -DPIN_SERIAL2_TX=PA_10''. If you dont use TFT pin you must also change the pin number you used. (PA_10 etc.)

brandon-northcutt, 2023-11-17:
Wow! I just lost a whole day thinking my CR-10s display was somehow incompatible with this SKR 3 EZ board. Then I found the serial coms were not coming up either. The problem was indeed that STM32H723Vx_btt will build fine but does not work with the STM32H723VG_btt on the SKR 3 EZ. Twitched to STM32H723VG_btt and everything works. BTT, please update your user manual!