bigtreetech/SKR-3 #54: Z axis and Extruder not working

GeniusPsychopat, 2023-04-09:
I got a new SKR 3 (not EZ), and after some tests i found out that the Z axis and the extruder aren't working. By some more testing with a spare motor and cable, I determined that the problem is on the board. I was also trying to run dual Z axis supplied by creality, which had a spliter cable instead of another 1in1out cable as I thought. On the first run i did accidentatly plug the extruder into ZMB, but i don't think that would break it like that. The config is salvaged from SKR 2 i had a while ago, but I did modify it to suit the new board.

GeniusPsychopat, 2023-04-09:
With a bit of testing with a spare motor I concluded that one of my TMC2208 was acting up, but that only accounts for the Z axis. When i tried to put the driver onto the extruder, it only hummed, I just hope that the board is fine. When I took the bad driver off of the board it didn't move though. Makes me wonder if there is something wrong with the stock A4988 coming built in the board.

radek8, 2023-04-11:
Check the connection of the coils on the motor connector.
There are two variants of connecting the coils in a stepper motor.

radek8, 2023-04-13:
SKR has one connection.. The engines are in two variants. 
[image]

GeniusPsychopat, 2023-04-13:
The problem was probably in my code, I finally got someone to look at it and the E0_RSENSE was set incorrectly. But now the board refuses to boot up. It's silent with the screen lit up, not showing anything.