bigtreetech/SKR-3 #55: unknown command m420.. When trying to home.

pnelson29, 2023-04-16:
I am not able to get my machine to home. it throws an Unknown Command M420. 
All I can find is about bed leveling.. I have that disabled as I don't have any bed leveling hardware

dgknow, 2023-05-04:
Hi pnelson29.
I am having the same issue with my SKR Mini E3 V3.0 board. Were you able to find the root cause and resolve the issue?

pnelson29, 2023-05-04:
No, I  still get it every time I home any axis.

dgknow, 2023-05-04:
I found this previous thread;
https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/2747

It looks like an issue with the TFT35 display firmware. I'm now looking  for a different firmware to install on my display.
 I'll let you know when I find a firmware that works.

I hope this is helpful.

fjbold1, 2023-08-10:
Same problem here. I'm working on it.

fjbold1, 2023-08-10:
It has been solved with a new update from bigtreetech, dated 8/10/2023.

dgknow, 2023-08-15:
Thanks fjbold1!
I'll try it out when I get time.

pnelson29, 2023-08-16:
> 

What update did you find? And how did you apply it? Sorry noob here

fjbold1, 2023-08-18:
search github for Bigtreetech tft touchscreen firmware, you can download the zip fie, it covers all models.
BIGTREETECH-TouchScreenFirmware-master (1).zip"