bigtreetech/SKR-3 #57: No Printer Attached Issue

Donovan4097, 2023-04-23:
Hello, I am having the no printer attched issue with my printer. I have an Ender 5 plus with an SKR 3 EZ and a TFT50 V3.0.1. I will provide my configuration.h below but I have followed the manual, and I have gone through every no printer attched post I can find. Nothing helps, my current settings for the tft are Serial Port = 1, Serial Port 2 = -1, Serial Port 3 = 3 as this is what the manual reccomends. I have tried every combination that I can think of but maybe I am missing something. I would appreciate some help. It wont let me upload the .h file so here is the .txt file for config
[Configuration.txt](https://github.com/bigtreetech/SKR-3/files/11302666/Configuration.txt)
[Configuration_adv.txt](https://github.com/bigtreetech/SKR-3/files/11302668/Configuration_adv.txt)

radek8, 2023-04-25:
Do you have the same serial speed set?
Did you use the correct coneltor?
Have you checked the TFT cable to see if it is faulty?

Donovan4097, 2023-04-25:
I used the TFT connector and I ohmed the cables, they are good. and yes I have the serial speed set to 115200 on the screen and marlin. Im using marlin 2.0.9.3 does this matter? I read something that says I may need marlin 2.1 or higher.

radek8, 2023-04-25:
Yes 2.0.9.3 is an old version, but it should not prevent the display from communicating with the SKR.
At most, some functions will not be available.

Donovan4097, 2023-04-25:
I just finished building marlin 2.1.2 for the skr 3 ez. I used a premade firmware from btt but marlin has a built in skr 3 ez board. lets see if it works

Donovan4097, 2023-04-25:
Update: new firmware didn't solve anything. Still have no printer connected.

Tomzrk, 2023-04-27:
Mine wasn't working either, i had the wrong envs selected in Platformio

[platformio]
src_dir = Marlin
boards_dir = buildroot/share/PlatformIO/boards
default_envs = STM32H743Vx_btt
include_dir = Marlin

Instead it should be

[platformio]
src_dir = Marlin
boards_dir = buildroot/share/PlatformIO/boards
default_envs = STM32H723Vx_btt
include_dir = Marlin

Donovan4097, 2023-04-28:
But there is no such thing as STM32H723 only 743

Tomzrk, 2023-04-28:
Are you using the newest marlin ?

Donovan4097, 2023-04-28:
Yes Marlin-Bugfix 2.1.x

Donovan4097, 2023-04-28:
[image]

Tomzrk, 2023-04-28:
First you need to fix the motherboard in configuration.h 
It should be for skr 3ez not regular 

#define MOTHERBOARD BOARD_BTT_SKR_V3_0_EZ

Tomzrk, 2023-04-28:
Than you go to platformio and change the line i sent you and try to build the firmware and try it

Donovan4097, 2023-04-28:
[image]

Tomzrk, 2023-04-28:
Can you show me the valid names

Donovan4097, 2023-04-28:
[image]
Error: Unknown environment names 'STM32H723Vx_BTT'. Valid names are 'BIGTREE_GTR_V1_0, at90usb1286_dfu, FYSETC_E4, STM32F103VE_ZM3E4V2_USB, mks_skipr_v1_nobootloader, STM32G0B1RE_btt, mks_robin_lite3, STM32F103VE, STM32F103RE_creality_maple, STM32F103RE_creality, STM32F103CB_malyan_maple, STM32H743Vx_btt, teensy35, simulator_linux_debug, FYSETC_S6_8000, sanguino1284p, mks_robin_nano35, STM32F103RC_btt_512K_maple, STM32F103RC_btt, trigorilla_pro, jgaurora_a5s_a1_maple, at90usb1286_cdc, teensy36, Opulo_Lumen_REV3, mks_robin_e3p_maple, mks_monster8, STM32F103VC_ZM3E4_USB_maple, DUE, LERDGEX_usb_flash_drive, STM32F103RC_maple, mks_robin_nano35_maple, STM32F407VE_black, mks_monster8_usb_flash_drive, teensy41, BIGTREE_OCTOPUS_V1_USB, STM32F103CB_malyan, mks_robin_lite3_maple, mks_robin_nano_v1v2_usbmod, LERDGES, BIGTREE_OCTOPUS_PRO_V1_F429_USB, mks_robin_e3, LPC1768, SAMD51_grandcentral_m4, BIGTREE_SKR_2_F429_USB, STM32F103VE_longer, Artillery_Ruby, chitu_f103_maple, STM32F103RC_btt_USB_maple, FLYF407ZG, STM32F103RC_fysetc_maple, 
trigorilla_pro_disk, LERDGEK_usb_flash_drive, mks_eagle_usb_flash_drive, melzi_optiboot, rambo, STM32F103VE_GTM32, mks_robin_nano_v1_3_f4_usbmod, BIGTREE_OCTOPUS_V1_F407_USB, STM32F103VE_ZM3E4V2_USB_maple, FYSETC_SPIDER_KING407, mks_robin_nano_v1v2, PANDA_PI_V29, melzi_optimized, mks_robin_lite, mks_robin_e3p, Opulo_Lumen_REV4, FYSETC_S6, sanguino644p, BIGTREE_SKR_2, mks_robin_pro, STM32F103RET6_creality_maple, mega1280, mks_skipr_v1, mks_robin_maple, BTT_SKR_SE_BX, mks_robin_nano_v3_1_usb_flash_drive_msc, mega2560ext, sanguino1284p_optimized, mks_tinybee, mks_robin_lite_maple, mks_robin_nano_v3_usb_flash_drive, mks_ks_robin_nano_v3, STM32G0B1RE_btt_xfer, mega2560, mingda_mpx_arm_mini, STM32F103RC_ZM3E2_USB, STM32F103RE_btt_maple, malyan_M300, STM32F103RC_ZM3E2_USB_maple, MightyBoard1280, BIGTREE_SKR_2_F429_USB_debug, mks_robin_e3_maple, STM32F401RC_creality_stlink, simulator_macos_debug, jgaurora_a5s_a1, mks_robin_pro_maple, STM32F103RC_creality, STM32F401RC_creality_jlink, STM32F401RC_btt, esp32, mks_monster8_usb_flash_drive_msc, STM32F103RE_creality_xfer, STM32F103VE_longer_maple, simulator_windows, chitu_f103, BIGTREE_SKR_PRO_usb_flash_drive, STM32F401RC_creality, trigorilla_pro_maple, simulator_macos_release, mks_robin_mini, STM32F103RE_maple, linux_native, STM32F103RE_btt, STM32F103VE_GTM32_maple, STM32F446_tronxy, LERDGEK, STM32F103VC_ZM3E4_USB, mks_robin_nano_v1v2_maple, TH3D_EZBoard_V2_no_bootloader, STM32F103RC_fysetc, FLY_MINI, MightyBoard2560, SAMD21_minitronics20, STM32F103RET6_creality_xfer, chitu_v5_gpio_init_maple, BIGTREE_E3_RRF, mks_eagle_usb_flash_drive_msc, chitu_v5_gpio_init, PANDA, include_tree, melzi, FYSETC_CHEETAH_V20, flsun_hispeedv1, STM32F103RET6_creality, STM32F103RC_meeb_maple, STM32F103RC_btt_512K_USB_maple, FLY_MINI_maple, BIGTREE_BTT002, Anet_ET4_no_bootloader, STM32F103RC_meeb, STM32F103RC_creality_xfer, BIGTREE_BTT002_VET6, STM32F103ZE, STM32F103RE_btt_USB_maple, STM32F103RE_creality_smartPro, STM32F103RE_btt_USB, mks_robin_nano_v3_usb_flash_drive_msc, BIGTREE_SKR_PRO, simulator_linux_release, STM32F070RB_malyan, Anet_ET4_OpenBLT, STM32F103RC_btt_512K, STM32F070CB_malyan, BIGTREE_SKR_2_F429, LERDGEX, STM32F103RE, BIGTREE_OCTOPUS_PRO_V1_F429, STM32F103RE_creality_smartPro_maple, mks_robin_nano_v3_1_usb_flash_drive, DUE_USB, STM32F103VE_creality, FYSETC_F6, ARMED, ERYONE_ERY32_MINI_maple, LPC1769, teensy20, BIGTREE_SKR_2_USB_debug, BIGTREE_SKR_2_USB, DUE_archim, melzi_optiboot_optimized, BIGTREE_OCTOPUS_V1_F407, mks_robin_nano_v3_1, STM32F103RC_btt_USB, rumba32, LERDGES_usb_flash_drive, teensy31, BIGTREE_OCTOPUS_V1, mks_robin_nano_v1_3_f4, BTT_EBB42_V1_1_filament_extruder, mks_robin'

Tomzrk, 2023-04-28:
Double check the number on the board chip

Donovan4097, 2023-04-28:
Im about to jump off a bridge, downloaded bugfix for the 3rd time tonight and now 723 shows up

Tomzrk, 2023-04-28:
Make sure your board chip end with 723 too mine does

Donovan4097, 2023-04-28:
THank you for your patience but that was wild

Tomzrk, 2023-04-28:
No worries, I been through the same thing just trying to help with my little knowledge

Donovan4097, 2023-04-28:
TFT is now connecting to my skr 3 ez, you are a lifesaver. thank you for all of your help.

Tomzrk, 2023-04-28:
You're welcome

giri456, 2023-12-31:
hello i also have the same issue except that my chip matches the environment in platform .io i also cant get the marlin mode working

radek8, 2023-12-31:
@giri456
Your description of the problem is too general.
Write what HW you have, describe exactly your problem.
Send the configuration files for SKR and TFT that you used.
Send photos of the cable connection between SKR and TFT

giri456, 2023-12-31:
@radek8
Sorry about that, first time on the forums. 
I am using a skr 3 ez and trying to connect it to a tft24 v1.1 in both marlin and touchscreen mode

For the tft i used the files found [here](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware) I directly coppied and pasted the ones that were for the tft24. 
[configuration_adv.txt](https://github.com/bigtreetech/SKR-3/files/13801404/configuration_adv.txt)
[configuration.txt](https://github.com/bigtreetech/SKR-3/files/13801405/configuration.txt)

giri456, 2023-12-31:
Sorry forgot about the photo.

[image]
[image]

[image]

radek8, 2023-12-31:
Also send the config.ini file, with your edit

giri456, 2023-12-31:
i used the default config.ini that is on the big tree tech github for this screen

radek8, 2023-12-31:
The default communication speed of the serial port is 115200 for TFT
Your Marlin configuration is 250000
You must have the same baud rate configured.

radek8, 2023-12-31:
In Marlin, enable the option
define M115_GEOMETRY_REPORT

giri456, 2023-12-31:
made the changes and did them on a fresh copy of marlin 
[configuration.txt](https://github.com/bigtreetech/SKR-3/files/13801698/configuration.txt)
[configuration_adv.txt](https://github.com/bigtreetech/SKR-3/files/13801699/configuration_adv.txt)

giri456, 2023-12-31:
[image]
[image]

giri456, 2023-12-31:
Got a few more pictures of what happened in marlin mode
[image]
[image]

giri456, 2024-01-01:
@radek8 
#define ST7920_DELAY_1 DELAY_NS(300)
#define ST7920_DELAY_2 DELAY_NS(300)
#define ST7920_DELAY_3 DELAY_NS(300)

I also tried adding this into config.h still doesnt work

giri456, 2024-01-01:
@radek8 i managed to get the touch mode to work i think it looks like this now

giri456, 2024-01-01:
[image]

giri456, 2024-01-01:
But Marlin mode is still not working
[image]

giri456, 2024-01-01:
[image]

radek8, 2024-01-01:
How long are your EXP1 and EXP2 cables? It looks like electromagnetic interference on the cables

radek8, 2024-01-01:
If Touch Mode works for you, you don't need Marlin Mode. 
I haven't used Marlin mode in over a year.

giri456, 2024-01-01:
@radek8 
[image]
I have uploaded a picture of the length of the cable. To confirm I can use a sd card on the tft with just the touch mode cable?

radek8, 2024-01-01:
>To confirm I can use a sd card on the tft with just the touch mode cable?

Yes, you can use the SD card on the TFT with a serial (5pin) cable.
If you set in Marlin:
#define SDCARD_CONNECTION ONBOARD
Then the mini SD card on the SKR will also be available in touch mode.

giri456, 2024-01-01:
Ok thanks

giri456, 2024-01-06:
@radek8 i finished my config.h and adv.h. i just tried flashing the firmware today and the file doesnt get renamed. the tft says eeprom data size error

radek8, 2024-01-06:
Did you compile for the right platform? 
After updating the marlin, you should initialize the EEPROM

giri456, 2024-01-07:
@radek8 I have compiled to the right processor in platform.io

radek8, 2024-01-07:
Are you updating Marlin?.
Use the microSD slot on the SKR.  
Use an SD card that is 8GB or smaller. 
How big is your BIN file? 
What is the designation of your processor on the SKR?

giri456, 2024-01-08:
Turns out it was a sd card problem. But now after flashing my z2 stepper won’t move or lock. It’s like it’s not getting power.

giri456, 2024-01-08:
[config_adv.h.txt](https://github.com/bigtreetech/SKR-3/files/13860893/config_adv.h.txt)
[config.h.txt](https://github.com/bigtreetech/SKR-3/files/13860894/config.h.txt)

giri456, 2024-01-08:
nvm got everything working much thanks @radek8

radek8, 2024-01-08:
I see that you don't have a USB port activated in Marlin. Are you going to use the USB port on the SKR?