bigtreetech/SKR-3 #62: SKR fails USB connect after sometime

yasaslive, 2023-05-26:
All SKR boards I used failed on USB connection after using sometime. This issue present in all SKR models including SKR1.4 / PRO SKR3/PRO/EZ. No apparent cause found. No any connection whatsoever to raspberry. Thinking leaving BTT for MKS.

MagicPiciu, 2023-05-28:
Same problem with me. Now I am looking for another board from another brand. After several searches on the Internet, I saw that it happens quite often with their motherboards.

yasaslive, 2023-05-28:
What is you firmware Klipper/Marlin?

MagicPiciu, 2023-05-28:
Klipper

yasaslive, 2023-05-28:
If you are using Fluidd Try Mailsail! Some are having issues with Fluidd and it's deprecated!

MagicPiciu, 2023-05-28:
I used Fluidd, Mainsail, Octoprint. It's the same for all: 
Klipper reports: ERROR
mcu 'mcu': Unable to connect
Once the underlying issue is corrected, use the
"FIRMWARE_RESTART" command to reset the firmware, reload the
config, and restart the host software.
Error configuring printer

yasaslive, 2023-05-28:
Yes, Exactly! I'm trying to use can over USB or UART for now until a new board arrives!

MagicPiciu, 2023-05-28:
I'm trying to connect to CAN but I can't. I don't know what else to do.
What motherboard did you order?