bigtreetech/SKR-3 #64: SKR3 EZ Sensorless Homing not working

asdino2, 2023-06-05:
motors don't stop at the end of bed. still running . Jumpers are on the board . If I connect the motor and give direction , will run 200, 300 400 mm even the bed is 200mm

radek8, 2023-06-06:
Did you compile your own FW?
How big did you make the bed?
Did you enable SW endstop?

asdino2, 2023-06-06:
Yes , I compiled on Marlin 2.1.2.1 . Bed is 245x 205 for a Prusa mk3 . Yes software end stop are enable . I put he jumper also . Your EZ 2209 seems not to see the overcurrent . I set up for 800 mA and nothing . Changed with 200 mA or 1000mA and still nothing Already I lost 1 EZ 2209 module .

radek8, 2023-06-07:
Do you use sensorless guidance?

radek8, 2023-06-07:
If you execute the command home (G28)
does the print head stop on the microswitch or are you using sensorless guidance and the head hits the frame?

asdino2, 2023-06-07:
The X or y hits the frame

radek8, 2023-06-07:
I see
I didn't notice the title
Sensorless Homing
So you are using sensorless guidance.
you have to fit the jumpers.

[image]

You need to enable guidance in Marlin and set the parameter for your engines

#define SENSORLESS_HOMING 
#define X_STALL_SENSITIVITY
#define Y_STALL_SENSITIVITY

asdino2, 2023-06-07:
I did already that , as I explained,  and is not working,  for this reason I write you . I follow  the manual . I put Mac sensitivity 250 and nothing. 
Thanks

radek8, 2023-06-07:
By changing the value
#define X_STALL_SENSITIVITY
#define Y_STALL_SENSITIVITY
you change the sensitivity. you need to find the correct value for your printer

asdino2, 2023-06-08:
ok, I will try every number from 0 to 255.I have 1 EZ2209 which I insert inside is overheating the chip and is not working. Can you replaced in warranty ?I have also ESP32U which in manual is write it is possible to flash the firmware buy the microSD. Is not possible , I tried and is not communicate with the motherboard 
Thank you

radek8, 2023-06-08:
I'm not from BTT. 
BTT does not answer here. 
You must contact the dealer to claim a defective part.

radek8, 2023-06-08:
Yes, you need to find the correct value between 0 and 255.
If the value is low or high, the engine will not spin or hit the frame..
I use microswitches. This feature behaved unreliably for me

asdino2, 2023-06-08:
Thank you for help me , I tried all the numbers from 1 to 255 and no end  stopping . Sensorless homing not working on skr3 ez for sure . I ask also other blogers on you tube and they did not test this function . they use end stop switch .what else I can do ? Is not working

radek8, 2023-06-08:
Do you have a diag jumper fitted?

radek8, 2023-06-08:
How are you on the impact sensitivity values?
Did you compile the FW with the new value or did you use the M914 code?

asdino2, 2023-06-08:
I use the m914 code from the screen and save it . After that tried and no response . I am sure the board don't  see the curent sensitivity

asdino2, 2023-06-08:
Yes, I have the jumper. I forget to tell you sometimes I get a error like "All low " when I power on the board, but not all the time , can be this a problem ? Do I need to set up auto home automatically?

radek8, 2023-06-08:
The "All low" message is a problem.
It seems that some TMCs have a problem with serial communication with the processor.
Send the M122 command and copy the response here.

Send configuration files Configuration.h and configuration_adv.h

ReefWarrior3D, 2023-08-06:
Any update on this?

asdino2, 2023-08-07:
Still not working !!!

mateo928, 2025-01-07:
> Il messaggio "All low" è un problema. Sembra che alcuni TMC abbiano un problema con la comunicazione seriale con il processore. Invia il comando M122 e copia la risposta qui.
> 
> Invia i file di configurazione Configuration.h e configuration_adv.h
> 
> Quale TMC hai installato? Puoi inviare una foto?

hi raden8, can you help me? i have the same problem