bigtreetech/SKR-3 #66: Configuration of drivers on SKR 3 EZ with five EZ5160 Pro drivers for separate Z-axis motors

BahamutWolf, 2023-06-11:
Hello BigTreeTech Team,

I recently purchased the SKR 3 EZ mainboard along with five EZ5160 Pro drivers. My goal is to connect and control two separate Z-axis motors independently. I have one driver each for the X-axis, Y-axis, ZA-axis and ZB-axis, E0 extruder, and E1 extruder.

Could you please provide me with guidance or instructions on how to properly configure the EZ5160 Pro drivers and the corresponding ports on the SKR 3 EZ to achieve separate control of the ZA and ZB axis motors? Are there any specific settings in Klipper that need to be adjusted to enable this configuration?

Thank you in advance for your assistance!

Mubbler, 2023-06-12:
za and zb are on the same driver. You have to sacrifice an extruder for a dual independent z motors.

BahamutWolf, 2023-06-13:
> za and zb are on the same driver. You have to sacrifice an extruder for a dual independent z motors.

That's what I thought, but it's not a problem because I only have one extruder. How do I have to set this in Klipper so that it treats E1 as the 2nd Z axis? Thanks very much.

Mubbler, 2023-06-13:
just change the klipper header of [stepper_extruder1] to [stepper_z1] and the same thing with the driver header. You might need to switch the direction of the the second z motor by placing "!" in front of the pin number for direction (i.e. dir_pin: PF12 to dir_pin: !PF12) and also change the run current to match the first z  motor. 

you will need to add [z_tilt] to level the surface prior to each print.

BahamutWolf, 2023-06-13:
> just change the klipper header of [stepper_extruder1] to [stepper_z1] and the same thing with the driver header. You might need to switch the direction of the the second z motor by placing "!" in [...]

Many thanks for the help. I will try it as soon as everything is there and installed. Thanks again

closed 2023-06-25 by BahamutWolf