bigtreetech/SKR-3 #70: SKR3 + Mini 12864 V2.0 Marlin config

dB-0103, 2023-06-22:
Can someone possibly point me in the right direction? I recently purchased an SKR 3 board and Mini 12864 V2.0 display with the intention of using them together...

I've been playing with the Neopixel configuration settings in Marlin for hours and I can't for the life of me figure out how to get the LEDs to work correctly. They turn on but when you try to edit the colors from the firmware menu after flashing, the colors don't match up with the options and the two LEDs under the encoder are always two different colors.

I've concluded that #NEOPIXEL_TYPE needs to be set to NEO_GRB (as shown in the sparse documentation for this display) but when you try to compile Marlin in this way, it fails and tells you the the type needs to be NEO_RGB which is clearly incorrect because it does not work correctly.

thisiskeithb, 2023-06-30:
`NEO_RGB` is the correct type for this display, but colors look odd on the SKR 3 series. I originally thought this was something STM32H7-specific, but colors look correct on the Octopus Max EZ with the same series of processor.

As for the solution... 🤷

dB-0103, 2023-07-04:
So that really makes me wonder... There must be some sort of bug in Marlin. The incorrect colors only happen when I try to edit the color from the `LED_EDIT_MENU`. If I have a specific color to set at boot using `LED_COLOR_PRESETS` it works fine, and similarly, `PRINTER_EVENT_LEDS` sets the correct colors. Very strange indeed. In any case I don't need to be able to change the color all the time, was just wondering if there was a solution or not. Thanks.