bigtreetech/SKR-3 #75: FAN or MFAN?

BahamutWolf, 2023-07-23:
Hello,

I know this is probably a rather stupid question but the SKR 3/EZ has FAN1, FAN 2, FAN 3 and the 3 MFAN connections. I wonder now but do I have to connect the Hotendkühler and my two component fans to the FAN 1- 3 connections or to the MFAN connections? Unfortunately I could not really find anything on the internet. In a video had it one at the MFAN connections but that seems to me somehow not right and I do not want to shoot me later the board. I had so far only with PC motherboards zutun. What exactly are the MFAN connectors for?

FwedFwed, 2023-08-27:
Hi,

I'm also interested. Anyone knows?

Schmudus, 2023-08-27:
Any news?
MAybe they can be used for cooler fans for stepper motors?

FwedFwed, 2023-08-29:
I found this thread on the BTT reddit : https://www.reddit.com/r/BIGTREETECH/comments/yq7rlf/using_vbb_fan_connectors_on_skr_3/
I think it may answer your questions.