bigtreetech/SKR-3 #77: SKR 3 EZ HE0 & HE1 jumpers

FwedFwed, 2023-08-27:
Hello,

I haven't found any documentation on the HE0 and HE1 jumpers located near the V-FAN jumper/under the 3 CNC fan connectors.

Does anyone know what they are for? I'm planning to run dual z with two drivers (Z and E0, E1 will be used for the extruder).

Thanks

Schmudus, 2023-08-27:
I don't know for what they are but I am using dual Z-Motor (I use Z and E1, E0 for extruder, leave this jumper as they are. Works very well :-))

Danez-DK, 2023-09-28:
It's for disabling the 5V -> 12V/24V mosfet for the HE0/HE1 connectors.
I think the jumpers just are the 5V bridge to the mosfets, so you proberly would have to connect to one of the jumpers pins to use the 5V PWM for a laser/fan, i have tried removing them, and use the HE0/HE1 connectors, but it just disconnected them.
Next step is to try one of the jumper pins, just have to get the right size duponts for it first.