bigtreetech/SKR-3 #85: Hardware (SKR 3 EZ)/BIGTREETECH SKR 3 EZ V1.0 3D.pdf is blank

jimmyeao, 2024-01-12:
Trying to look at the manual for the BIGTREETECH SKR 3 EZ V1.0 but the pdf is blank:
[image]

thisiskeithb, 2024-03-24:
[BIGTREETECH SKR 3 EZ V1.0 3D.pdf](https://github.com/bigtreetech/SKR-3/blob/master/Hardware%20(SKR%203%20EZ)/BIGTREETECH%20SKR%203%20EZ%20V1.0%203D.pdf) is an interactive PDF with multimedia/3D content that cannot be viewed through your browser. Download the PDF and view in Acrobat Reader directly:

[image]

All of BTT's PDFs with "3D" in the name will be like this.

jimmyeao, 2024-03-25:
Found it, the manual is located in the wrong place in the directory tree:
[image]
Its located inthe stander SKR 3 section not the EZ section

closed 2024-03-25 by jimmyeao